fritzing/fritzing-app #3643: Qt versions being used for builds

mMerlin, 2020-03-14:
## Problem

Setting up an environment to use Qt Creator, I have been browsing around the available application build information.  From what I found, some of the documentation has not been update to the the current practises.

[Building-Fritzing](https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing) on the wiki says …

> We are always trying to use the latest version of Qt. With Fritzing 0.9.3 we updated to Qt 5.6. With Fritzing 0.9.4 we are updating to Qt 5.9.6.
>
> The online installer is the easiest route to go. In the installer, you need to pick the right version (5.12) and the compiler for your platform:

Then [Publishing a Release](https://github.com/fritzing/fritzing-app/wiki/4.-Publishing-a-Release) says

> We currently use Qt 5.6 for builds on all platforms

## Proposed Solution

Get a consistent version across the pieces of documentation, or qualify each reference with the Fritzing version that it applies to.

I **ASSUME** that 0.9.5 is actually using 5.12, although that does not quite agree with the `latest version` reference.  5.14 is currently available through the online installer.

mMerlin, 2020-03-14:
Unless I am using the wrong version of Qt sdk and creator, the run instructions at the end of [Building Fritzing](https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing) could use a few more details.  At least for first time use and Fedora 31.  I could edit that myself, but figure some confirmation from someone that is (and has been) using the described environment would be good first.  To find out if it is just my Fedora environment, or choice of versions that is causing the problem.

I was initially unable to follow the Qt Creator run instructions from the referenced wiki page. There is no **Run Settings** section, or Arguments field visible.

[image]

The `Desktop Qt 5.12.7 GCC 64 bit` entry under **Build and Run** is ghosted.  Hovering over that shows
> Kit is unsuited for project
> Error: No C++ compiler is set in kit.

Going to `Manage Kits » Kits » Auto-detected » Desktop Qt 5.12.7 GCC 64 bit (default)` shows

[image]

Going into **Manage** for compiler, I was able to set GCC as the C++ compiler, but for some reason it did not auto-detect that, though it found it for the C compiler.  With that updated, back on the projects page, I could activate the kit, which then shows build and run buttons.  The command line arguments are found under **Run**.

So a few extra steps are needed to get the project ready to build and run, and the field labels are slightly different.  For Fedora 31 at least.

KjellMorgenstern, 2020-03-14:
I corrected the Qt version statement in the release process. Other version statements seem correct to me.

closed 2020-03-14 by KjellMorgenstern