fritzing/fritzing-app #3644: dpi recommendation

KjellMorgenstern, 2020-03-14:
Fritzing needs the physical size of parts and graphics.
This is crucial in the PCB and Breadboard output for print or etching. It is also important for correct rendering and having parts at the same scale. 
Guessing which dpi a graphic was designed in might be convenient, but it does more harm than good.

The best way to do that is to specify graphics height and width in inches or mm.
Therefore
[ ] adapt all recommendation on the wiki 
[ ] adapt all recommendation on the webpage
[ ] replace template files for creating parts
[ ] add instructions how to configure "inch" or "mm" in Illustrator 
[ ] add instructions how to configure "inch" or "mm" in Inkscape
[ ] add a warning in the parts checker for core

The general recommendation to use 1 as "one mil" (meaning 1000dpi) makes sense. There is nothing against using 1 as 0.1mm (254dpi) or similar. But this independent from the recommendation to specify the image size. Using lower resolution 96 dpi is fine, as long as the unit of width and height is given in milimeter or inches.

mMerlin, 2020-03-16:
Add to that updating/adapting existing examples, though since the core parts are often pointed to as "examples" that will be ongoing for awhile.

deshipu, 2020-10-01:
I was just bitten by Inkscape's change of default DPI — designed my board outline in Inkscape, imported it, and didn't notice it was roughly 10% larger. Then I was a bit surprised when the PCBs came — anybody wants a keyboard PCB for someone with really large hands?