fritzing/fritzing-app #3647: design rules check not seeing all of the copper for specific part

mMerlin, 2020-03-17:
## Current Behaviour

Fritzing design rules check is not detecting collision between traces and the elongated portion of the pads for the LED Matrix parts I added to core recently.  It does detect collisions with identical looking pads in the MAX7219 and sparkfun 556 time chips.  I am not seeing an obvious (functional) difference between the pcb image implementations.

The working chips:
* copper1 and copper0 groups separate (so duplicate connector«n»pad elements)
* path element that extends the copper pad immediately after each pad circle
* a portion of the path element uses an absolute move
* no transform attributes
* stroke, stroke-width, fill attributes are on each element
* the silkscreen is created from a series of line elements plus a single path
* the pads are (mostly) outside of the silk screen

LED Matrix with narrow pads:
* copper0 nested inside of copper0
* extending path elements in a group after connectorpad circles (but still inside copper1 and copper0 groups)
* only relative coordinates are used in the path (all of the copper paths are identical, except for the initial move to position them)
* outer copper1 group has a translate to align the first pad center to 0,0
  * group of extended copper paths has a translate to adjust the path origin to 0,0 (so the paths have the same m«ove» coordinates as the circles have x and y)
  * the second row of pads (and extended copper) have a translate to move over 600mils
* stroke, stroke-width, fill values are set by a style attribute on the containing group
* the silkscreen is created from a single rect element
* the pad are completely in side the silkscreen

In the screen shots, the top row of pads is for the LED Matrix.  The bottom row, left side is a MAX7219, the right is a 556 timer chip.  From the **red** areas, frc is only detecting the pad circle element as a collision candidate for the Matrix, not the oblong pad around it.

testing sketch file (renamed)
[design rules check research-01.fzz.zip](https://github.com/fritzing/fritzing-app/files/4341111/design.rules.check.research-01.fzz.zip)

0.9.4b collision screenshot
[image]

0.9.5 collision screenshot
[image]

So this says there is some combination of:

* a bug in the design rules check code about what it sees as (top and bottom) copper
* a real problem with the LED matrix part that I can not see
* a need for a test case when importing new parts, to verify drc can see all of the copper
* a need for the part validation script to detect invisible copper
* a need for documentation notes with more information about how to create valid pcb view images

Before I start making the LED matrix PCB image closer to the chips, to find out where the effective difference is, I figured it would be a good idea to report the problem, and see if other eyes/knowledge could narrow the scope.

**Build:**
Fritzing version 0.9.4.2019-12-01.CD-498 - Qt version 5.13.2
Version 0.9.5 (dCD-548-1-gcc8758a1 2020-02-15) 64 [Qt 5.13.2]

**Operating System:**

Fedora 31

**Steps to reproduce:**

Run design rules check on the sample file, and see where the collisions are detected

## Expected Behaviour

The design rules checker should detect collisions with copper (coloured) elements in valid parts.

KjellMorgenstern, 2020-03-17:
Does this also fail if the collision is only at the oblong pad? Meaning, DRC would not report any collision at all?

mMerlin, 2020-03-17:
Yes.  That is where I saw it first.  The example here was created for documentation purposes, to show the differences depending on the specific part.

mMerlin, 2020-03-18:
@KjellMorgenstern  Since I finally have a working qtcreator environment and Fritzing build on a vm, I am looking into this one.  So far, I have confirmed, through the existing debug png saves, that the current code is not seeing the extended pad areas for the problem part.  I have not figured out yet how to trace the svg content that is included in the copper.  In the investigation, I am learning a bit about how the drc code works.  Do you accept pull requests that are only comments added to the code describing its functionality?  The existing code is **VERY** light on descriptive comments, and anything I learn could be of use to someone else that needs to look at the code.  Even if I do not find the spot to fix the problem.

mMerlin, 2020-03-19:
For a sample, I told doxygen to generate svgs for the diagrams.  Here is what it created for the call graph of DRC::start_aux().  Renamed to append .zip to the name as usual.
[start_aux_call_graph.svg.zip](https://github.com/fritzing/fritzing-app/files/4356699/start_aux_call_graph.svg.zip)
The generated htlm includes tools to zoom and scroll around, without needing to save or view on a separate tab.

mMerlin, 2020-03-22:
It looks like the problem is not in the drc code itself.  Once I figured out enough of Fritzing and Qt data structures to find places to look at the Data, I started adding debug code (Frizting 0.9.5), to get a look at the data being processed.  I have still not figured out how to manually (in the debugeger) drill down through the data structures to the *interesting* data, at least most of the time.  But, in DRC::startAux (autoroute/drc.cpp)

```cpp
masterDoc->setContent(master, &errorStr, &errorLine, &errorColumn)
```

loads the information later used by DRC to generate the collision detection images.  After that was done, I added (with appropriate includes)

```cpp
std::ofstream dbgout((FolderUtils::getTopLevelUserDataStorePath() + QString("/master%1_Doc.svg").arg(viewLayerPlacement)).toStdString());
dbgout << masterDoc->toString().toStdString();
dbgout.close();
```

For the max7219 part, the displayed svg files looks correct, but the led matrix do not.

Snapshots of image viewer showing Max7219
[image]
[image]
Snapshots of image viewer showing LED Matrix
[image]
[image]

Having a look at the svg files, the path elements for the max7219 look correct, but the path elements for the led matrix are missing data.  The "d" attributes are empty strings, there are no other attributes.

From MAX7219

```xml
<path stroke-width="0" xmlns="http://www.w3.org/2000/svg" stroke="none" d="m54,300a26,26,0,0,1,26,26l0,52a26,26,0,0,1,-26,26l0,0a26,26,0,0,1,-26,-26l0,-52a26,26,0,0,1,26,-26l0,0zM38,352a16,16,0,1,0,32,0,16,16,0,1,0,-32,0z" fill="black"/>
```

From LED Matrix

```xml
<path d=""/>
```

`master` is populated from

```cpp
QString master = m_sketchWidget->renderToSvg(renderThing, m_board, viewLayerIDs)
```

Which is from sketch/sketchwidget.cpp.  That seems to be where the problem originates.
Continuing research from there.

KjellMorgenstern, 2020-03-22:
Starting the part (d=...) with a small “m” is a bit suspicious, as this SVG command is in relative coordinates. Maybe the parser is thrown of by that, as it expects the first command to be in absolute coordinates.

mMerlin, 2020-04-28:
The initial problem, not detecting the oblong expansion of the the connector pad, was due to a simple parse error. The svg specification for the path element (used for the oblong pad expansion) allows consecutive numbers to be delimited various ways.  Whitespace or commas are typical, but a negative number does not require any extra delimiter.  The minus character is the delimiter.  So "0-4" is equivalent to "0,-4" or "0 -4". Specifically for the the "a" (arc) command, the parser did not recognize the sequence. The tokenizer correctly saw that as "number number", but the parser was only accepting "number comma number" or "number space number". The parser code itself is (was) generated from a grammar file, which I found no documentation on. It results in several arrays of numbers that are used as combinations of indexes and offset in a convoluted piece of state machine logic. I could not determine how to fix the parser itself.  However, a simple patch was possible. There is an existing preprocesser for the tokenizer code that cleans up whitespace. Adding a step to that to insert a space before the minus characters does not change the sense of the path data, and allows the parser to handle the tokenized output.

Once that was implemented, a new drc processing error was uncovered. The oblong expansion was not being detected as attached to (part of) the connector pad, resulting in a bogus overlap report. That was (finally) traced to a small piece of code that was supposed to create that association dynamically. It failed for this case because the area of the bounding box for the connector pad was less than half the size of the bounding box for the oblong expansion. That code has been tweaked in #3664 to not care about that area ratio when the intersection of the bounding boxes is close to the size of the connector pad bounding box.

mMerlin, 2020-04-28:
The initial problem, not detecting the oblong expansion of the the connector pad, was due to a simple parse error. The svg specification for the path element (used for the oblong pad expansion) allows consecutive numbers to be delimited various ways.  Whitespace or commas are typical, but a negative number does not require any extra delimiter.  The minus character is the delimiter.  So "0-4" is equivalent to "0,-4" or "0 -4". Specifically for the the "a" (arc) command, the parser did not recognize the sequence. The tokenizer correctly saw that as "number number", but the parser was only accepting "number comma number" or "number space number". The parser code itself is (was) generated from a grammar file, which I found no documentation on. It results in several arrays of numbers that are used as combinations of indexes and offset in a convoluted piece of state machine logic. I could not determine how to fix the parser itself.  However, a simple patch was possible. There is an existing preprocesser for the tokenizer code that cleans up whitespace. Adding a step to that to insert a space before the minus characters does not change the sense of the path data, and allows the parser to handle the tokenized output.

Once that was implemented, a new drc processing error was uncovered. The oblong expansion was not being detected as attached to (part of) the connector pad, resulting in a bogus overlap report. That was (finally) traced to a small piece of code that was supposed to create that association dynamically. It failed for this case because the area of the bounding box for the connector pad was less than half the size of the bounding box for the oblong expansion. That code has been tweaked in #3664 to not care about that area ratio when the intersection of the bounding boxes is close to the size of the connector pad bounding box.

@KjellMorgenstern starting the d attribute with "m" and "M" are both valid. I think the only functional difference between them (for regular svg) is if the following command is a line, without the "l" or "L" introducer. "m" implies "l", while "M" implies "L".  So "m10,10 20,20" is equivalent to "m10,10l20,20" while "M10,10 20,20" is equivalent to "M10,10L20,20".

The initial move is based on the current coordinate system, and both the absolute and relative moves are from 0,0 of that coordinate system. Absolute and relative offsets from 0,0 are equivalent. Testing shows that the parser is handling both cases

failiz, 2021-03-18:
already released, close

KjellMorgenstern, 2021-03-19:
retested on 0.9.6 release, works fine now

closed 2021-03-19 by KjellMorgenstern