fritzing/fritzing-app #3648: voltage value of capacitors not initially showing in the part label

mMerlin, 2020-03-17:
## Current Behaviour

When a capacitor part is placed, it is initially configured to included the voltage in the part label.  According to the "display/values" menu for the part label.  However, the voltage is not actually being displayed.  Turning unchecking and rechecking the voltage in the menu does not have any effect.

This is only being documented for the schematic view, but a quick check says it happens in breadboard and pcb views as well.  Turning on the part label in breadboard view does not show the voltage, unless it has been modified in Inspector.  Here again, the voltage is initially set to display according to the display/values menu.  In pcb view, the voltage is not initially set to be displayed.  Turning it on does not display the voltage, unless it has been modified in the inspector.

Interesting side note.  I just discovered that I can take a screenshot bigger than my physical monitor.  In the VM, setting a screen resolution bigger than the host monitor means (one scenario) scrolling around to see the edges.  However, a screenshot of the current window (on the VM) gets the full defined area/resolution.

initial schematic  view version 0.9.4b
[image]

modified voltage of one part in inspector (and change back to initial value)
[image]

initial schematic  view version 0.9.5
[image]

modified voltage of one part in inspector (and change back to initial value)
[image]

**Build:**
Fritzing version 0.9.4.2019-12-01.CD-498 - Qt version 5.13.2
Version 0.9.5 (dCD-548-1-gcc8758a1 2020-02-15) 64 [Qt 5.13.2]

**Operating System:**
Fedora 31

**Steps to reproduce:**

- place a capacitor part
- examine the part label text
- examine the enabled fields for the text label
- modify the part voltage in Inspector
- see that that the voltage is displayed in the part label

## Expected Behaviour

The voltage should always be displayed when it is enabled in the display/value menu, and the label itself is enabled.

failiz, 2020-11-27:
I was programming a new feature and I saw that when the calling capacitorItem->getProperty("voltage") returns an empty string. (The inspector showed 6.3V). As soon as I change the value in the inspector, calling capacitorItem->getProperty("voltage")  returns the right value. So, similar issue as the one described here.

The issue seems to be that the voltage field is empty in the part definition. As soon as you write a value in the voltage field and regenerate the database, both issues are fixed. (calling capacitorItem->getProperty("voltage")  returns the right value and the voltage is displayed when you place the component.)

So, the easy fix is to fill some default values in the parts. The right fix would be to find the bug in the code. Interesting, the capacitors are initialized with a voltage of 6.3V, but in properties.xml the default value for capacitors is 35V...

KjellMorgenstern, 2020-11-28:
Could that be related to PR #3755 ?
I saw the code smell, but don't yet know the issues it might have caused.

failiz, 2020-11-29:
I do not think so. #3755 seems like an error, but the capacitor class is not based on SymbolPaletteItem class.

I made a PR that fixes the problem #3763

commit 9ee3b953 (2020-12-28): fixes #3648. If the property is empty, take the default one.
closed 2020-12-28 by KjellMorgenstern