fritzing/fritzing-app #3649: 60 Second delay on file open.

Ajb2k3, 2020-03-17:
**OSX 10.14.2
Fritzing 0.9.4**

**Issue:** 
When clicking the file open menu option, it takes 60 seconds for the dialog to be useable and then another 60 second delay if i click on the dropdown box's or the folder list.

KjellMorgenstern, 2020-03-17:
That seems to be a situation tightly coupled to your system, I would help to get as much information as possible. Also it is important to rule out other possible causes to narrow down the issue.

Possible checks:
Can you repeat this, does it happen each time?
I guess opening the same location in FInder (not with Fritzing) works fine, did you check? If yes, is there an unusual high number of files in that directory?
What does the system status say about CPU usage, is there a process taking a lot of CPU time?
Do you have any backups, indexers or virus scanner running?
Are you using network drives?

Ajb2k3, 2020-03-17:
**Possible checks:
Can you repeat this, does it happen each time?**

Yes, completely repeatable.

**I guess opening the same location in FInder (not with Fritzing) works fine, did you check? If yes, is there an unusual high number of files in that directory?**

No problem in Finder, Just Fritzing.

**What does the system status say about CPU usage, is there a process taking a lot of CPU time?**

Dunno, will check later but it is possible it is making out as other GUI programs stop working during the delay.

**Do you have any backups, indexers or virus scanner running?**

Nope
**Are you using network drives?**
and nope again.

KjellMorgenstern, 2020-03-17:
Is there an unusual high number of files in that directory?

Ajb2k3, 2020-03-17:
What do class as unusual?

Ajb2k3, 2020-03-21:
It's causing a gui lockup so don't often catch process details but often the TCCD process gets pulled to the top of the process list.

KjellMorgenstern, 2020-03-24:
About the TCCD process.
Fritzing should not access your contacts. Maybe it helps to remove this access right from Fritzing? See
https://zhuanlan.zhihu.com/p/25164713 for a brief summary , also fits your version of OS i think. 

Anyway this access right should not be there, as Fritzing does not request it. But let’s first check if that’s really what is going on, for now it is just a wild guess.

Ajb2k3, 2020-03-25:
Fritzing is also trying to access the microphone!

KjellMorgenstern, 2020-03-25:
How do you get to that conclusion, of course it doesn’t? No matter if it does or not, TCCD should not hog the CPU. 

What happens if you disable that access = remove the access right to contacts?

Ajb2k3, 2020-03-25:
Ok its gone from microphone but
[https://www.pinterest.co.uk/pin/497295983857290014](url)
[https://www.pinterest.co.uk/pin/497295983857290010/](url)

Ajb2k3, 2020-03-25:
Update,
Removing the permissions to everything but images has removed the delay.

KjellMorgenstern, 2020-03-25:
@Ajb2k3 The pintersest links are broken. 

Fritzing doesn't really need access to images either, it is not a mobile app. What it needs is file access to a "Fritzing" folder in the documents location, a place to store its config files and parts, and ideally network access (for projects, blog, updates etc)

KjellMorgenstern, 2020-03-27:
This doesn't seem happen normally, and a solution has been found. I am going to close this.

closed 2020-03-27 by KjellMorgenstern