fritzing/fritzing-app #3653: Duplicate components

NilsRo, 2020-03-30:
## Current Behaviour
Hi,

I have dublicate components which share same connections in vreadboard and circuit view.

[image]

**Build:**
CD-498-0-a1ffcea 2019-12-01 (Qt 5.12.3)

**Operating System:**
Windows 10

**Steps to reproduce:**
Happens sometimes.
## Expected Behaviour
No dublicate components

[PSU1 Adapterplatine.zip](https://github.com/fritzing/fritzing-app/files/4405860/PSU1.Adapterplatine.zip)

vanepp, 2020-04-15:
There in fact are no duplicate components. If in breadboard I select Routing->select all wires and then press delete (to delete all the wires) then switch to schematic and Routing select all traces and delete (to delete all wires) only the components are left:

[image]
[image]
[image]

so far so good, except there are more connectors in pcb than schematic or breadboard. However that is because Fritzing will load components on top of one another and you may have to drag them apart. 

[image]

Here the connector circled in red was placed on top of the connector circled in blue and will connect all its pins to the second connector creating unwanted connections. 

[image]

Similarly in schematic these three connectors (circled in red blue and black) were positioned on top of each other creating unwanted connections

[image]

Pcb was actually ok, other than the 1mm connector (circled in red) needs to have its pad sizes changed to .01 by .01 to make pads that do not over lap (the red circle in Inspector on the lower right of the screen.) There are no duplicate components only ones that were on top of each other. I'd probably argue that the multiple J1s are a bug (but I couldn't reproduce it the last time I tried) but in this case they are in different sizes of connectors and thus may be OK. 

Peter

NilsRo, 2020-07-26:
Hi. 

I now can reproduce the issue. If you use the mouse scrollwheel to change the type of a component fast the component will be dublicated with shared connectors (not visible but connected in the background). So if one of the components is deleted the original wires will be deleted as well. You can see that in the attached video. At the end all "new" components created has green connectors. So Also PCB, ... will be wrong.

[2020-07-26 11-40-51.zip](https://github.com/fritzing/fritzing-app/files/4977428/2020-07-26.11-40-51.zip)

[Sicherheitsschaltung Ladekoffer - V2-1 20200725.zip](https://github.com/fritzing/fritzing-app/files/4977430/Sicherheitsschaltung.Ladekoffer.-.V2-1.20200725.zip)

Cheers,
Nils

vanepp, 2020-11-01:
Yes, easily reproducable. Simplest case is a single rectifier diode. Drag a diode in to pcb, select the diode then in Inspector select the type field to produce the drop down menu, then select it again (which causes it to blank out, which is likely a bug as well!) like this:

[image]

then with the type field selected, scroll quickly with the wheel and new parts appear (which they shouldn't!)
 
[image]

 
What looks to be happening is the first change spawns a thread to make the requested change, but the scroll wheel is fast enough (unlike a usual human response of moving the mouse and clicking which is why we usually don't see this) to spawn a new thread to make another change before the first change has completed using the same moduleId. That appears to produce new instances of the same part (which may be the normal response) without deleting the original part as may normally happen. I think this causes internal routing corruption, because I have managed to hang Fritzing by making connections among the newly generated parts, and I suspect that all three are sharing the same data structure or possibly net names. One possible solution is to implement a busy flag (which has its own problems likely) that is set at the start of the first change and cleared when the change is completed. Then the later changes (no matter the timing) would wait til the busy flag is cleared indicating the firt change has completed before starting the next. This would imply a mechanism to queue requests til the busy flag is clear, but I expect there likely is such a mechanism in Qt. 

Peter

failiz, 2021-01-08:
This is a duplicate of #1431. Merge?
I think the best/easiest solution would be to disable the wheel for the properties and leave the wheel only for scrolling in the inspector pane

failiz, 2021-03-12:
Close in favor of #1431

KjellMorgenstern, 2021-03-12:
Thanks @vanepp and @NilsRo for reproducing the problem.

closed 2021-03-12 by KjellMorgenstern