fritzing/fritzing-app #3657: White note text

rr795, 2020-04-12:
In version 0.94 on mac you cannot change the color of the notes, the text is white and you don't see almost

rr795, 2020-04-12:
[image]

rr795, 2020-04-12:
You cannot change the color of the text to black, or to another color

theodore-dream, 2020-06-29:
I am adding a comment here to note that I am facing this issue and I've researched this and there is no workaround for this issue. Notes are impossible to see.

fritzing-0.9.2b-20.fc32.x86_64
Fedora release 32 (Thirty Two)
kernel version 5.6.14-300.fc32.x86_64

Best,
Richard

KjellMorgenstern, 2020-08-19:
Duplicate of #3530

domhardt, 2020-10-22:
The problem still exists and there is no solution for it AND it includes Fritzing's "About" dialogue.

Fritzing Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) Cocoa [Qt 5.13.2]
MacOS 10.15.7 (19H2) Dark Theme

[image]

[image]

KjellMorgenstern, 2020-10-22:
Yes, that why it is marked as solved (not closed) with the 0.9.5 milestone. It is fixed, but not yet released.

domhardt, 2020-10-22:
> Yes, that why it is marked as solved (not closed) with the 0.9.5 milestone. It is fixed, but not yet released.

Oh sry, I missed that. I'm waiting for the next release then. Thank you & keep on the good work!

KjellMorgenstern, 2020-10-22:
No problem, the "solved for milestone" concept is not used everywhere. I'll think about putting it in the README / contribution guidelines. And besides that, the 0.9.5 release really is overdue.

failiz, 2021-03-12:
Time to close all solved issues after the release of 0.9.6?

KjellMorgenstern, 2021-03-12:
Just retested, only partly solved. The "About" box still has the issue.

domhardt, 2021-03-13:
> Time to close all solved issues after the release of 0.9.6?

The release page still says that 0.9.4 is the latest stable release.

failiz, 2021-03-13:
@domhardt , 0.9.6 has been released on Feb 22. Where do you see that the latest version is 0.9.4?
https://blog.fritzing.org/2021/02/23/new-fritzing-release-096

domhardt, 2021-03-13:
> @domhardt , 0.9.6 has been released on Feb 22. Where do you see that the latest version is 0.9.4?
> https://blog.fritzing.org/2021/02/23/new-fritzing-release-096

Here:
[https://github.com/fritzing/fritzing-app/releases](https://github.com/fritzing/fritzing-app/releases)

blurpy, 2021-03-23:
I just tested 0.9.6 to see if the note text color was fixed, and it looks the same as 0.9.4, still white:

[image]

openSUSE Tumbleweed with KDE.

KjellMorgenstern, 2021-03-23:
Is this from an official build? We only tested this on Ubuntu and MacOS, however there are very few platform specific styles, so the issue still occuring on 0.9.6 is a bit odd, no matter which system.

blurpy, 2021-03-23:
I downloaded the source from here: https://github.com/fritzing/fritzing-app/releases/tag/0.9.6
And followed the instructions from https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing to build it.
Then started it from the build directory.

Maybe it's related to me using dark mode on KDE?

domhardt, 2021-03-23:
While I'm willing to test there is still no recent stable version available on the releases page of this repo. :(

blurpy, 2021-03-23:
Stupid me, when I started the application from the build directory, I forgot `./` in the command, so I actually started 0.9.4. (>.<)
Sorry for the confusion. It shows like this now:

[image]

KjellMorgenstern, 2021-03-23:
@blurpy Thanks for clarification

KjellMorgenstern, 2021-07-13:
Various fixes have been deployed in 0.9.6 and 0.9.7 (rc3) , marking this as solved.
If there are still issues, or new issues, please open a separate ticket with exact version of Fritzing, Operating System and Color Scheme settings.

closed 2021-07-25 by KjellMorgenstern