fritzing/fritzing-app #3658: Can't start

mattpackwood, 2020-04-14:
## Current Behaviour
Security on the Mac will not allow 0.9.4 open, but reinstalling 0.9.3b that will open...
**Build:**
Version 0.9.4

**Operating System:**
MacOS 10.15.4

**Steps to reproduce:**
Install and try to open

## Expected Behaviour
Text

KjellMorgenstern, 2020-04-16:
Instructions on how to start Fritzing on MacOS are on the download page:
https://fritzing.org/download/

vanepp, 2020-04-16:
Apparently a uninstall (using a cleaner of some kind) and reinstall fixed the problem. Don't know the original cause.

KjellMorgenstern, 2020-04-19:
MacOS Catalina has stricter security settings, especially for uncertified applications.This is the reason why Catalina is not officially supported by version 0.9.4. However there are quite many reports now that Fritzing works on Catalina, too. It would be interesting to see if there are additional steps required to start Fritzing, however I did not see any issue when trying in a Mac OS Catalina Virtual Machine.

closed 2020-04-19 by KjellMorgenstern