fritzing/fritzing-app #3662: but can Not run --- see error below...?

lse123, 2020-04-17:
## Current Behaviour
https://github.com/fritzing/fritzing-app.git
I CLONED THIS APP ^^^
in Qt Creator
but can Not run --- see error below...?
Also, I chose the MSVC2017 32-bit kit during cloning in Qt IDE--- how to add the 64-bit kit to this project...[If the project already imported]?

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>
posted and here:
https://forum.qt.io/topic/113561/also-i-choose-the-msvc2017-32-bit-kit-how-to-add-the-64-bit-kit-to-this-project

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour

2020-04-11T20:04:45 Clang Code Model: Error: The clangbackend executable "C:\Qt\Tools\QtCreator\bin\clangbackend.exe" does not exist.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_arm64_v8a/mkspecs/android-clang/qmake.conf.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_armv7/mkspecs/android-clang/qmake.conf.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_x86/mkspecs/android-clang/qmake.conf.
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2
Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
Project ERROR: libgit2 include path not found in F:/Qt-Projects/fritzing-app/../libgit2/include
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2
Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
Project ERROR: libgit2 include path not found in F:/Qt-Projects/fritzing-app/../libgit2/include
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2
Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
Project ERROR: libgit2 include path not found in F:/Qt-Projects/fritzing-app/../libgit2/include
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2
Project MESSAGE: See https://github.com/fritzing/fritzing-app/wiki for details.
Project ERROR: libgit2 include path not found in F:/Qt-Projects/fritzing-app/../libgit2/include
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_arm64_v8a/mkspecs/android-clang/qmake.conf.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_armv7/mkspecs/android-clang/qmake.conf.
Project ERROR: You need to set the ANDROID_NDK_ROOT environment variable to point to your Android NDK.
Could not read qmake configuration file C:/Qt/5.12.2/android_x86/mkspecs/android-clang/qmake.conf.

KjellMorgenstern, 2020-04-17:
Your build environment is not setup correctly. Building a release for Android might be a difficult task.

closed 2020-04-17 by KjellMorgenstern

lse123, 2020-04-17:
I mean release build for Windows 10 

I do not know about Android, forget Android --- is an error - of me - or may import from GitHub Not went well

I am interested in building for Win 10, your repo compiled is ok for both 32-bit and 64-bit same repo???

how add libgit2 ... to cy=urrent project?
Project MESSAGE: target arch: x86
Project MESSAGE: Fritzing requires libgit2
Project MESSAGE: Build it from the repo at https://github.com/libgit2

lse123, 2020-04-22:
well