fritzing/fritzing-app #3667: Connections between schematic and breadbord differ

Hamsterjh, 2020-05-02:
## Current Behaviour
Text
<please attach screenshot if possible>
[image]
[image]

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Mismatch of cct drawn in schematic view to the layout in breadboard view,
- Step 2

## Expected Behaviour
Text

KjellMorgenstern, 2020-05-02:
Where is the mismatch?

Hamsterjh, 2020-05-02:
As an example: 
Schematic - Capacitor C1 positive tied to 7805 positive input
Breadboard - Capacitor C1 positive shows dotted line to negative of 7805

Schematic - Capacitor C2 positive goes to pin 3 of 7805
Breadboard - Capacitor C2 shows no connection to pin 3 of 7805

Schematic - Capacitor C2 negative goes to common negative rail.
Breadboard - Capacitor C2 negative goes to positive input pin 1 of 7805 as well as negative rail, thus creating short circuit.

Thanks,

KjellMorgenstern, 2020-05-02:
In the schematic pin3 of 7805 is red, which means that it is not connected. 
You can also check by pressing the connection with the mouse, all connections should be highlighted.

7805 is numbered in the wrong order, this is issue https://github.com/fritzing/fritzing-parts/issues/252

The two red wires are not connected, there is no green circle around the connectors on the breadboard where they meet.

Hamsterjh, 2020-05-03:
Ok, thanks I did fix the unconnected pin 3 of the 7805, it shows green now.

Thanks for pointing out the pins order on the 7805, how come the pins of the 7805 is different to what is in the industry? Is there a fix planned for this?

KjellMorgenstern, 2020-05-03:
Honestly I don't know why this part is that way, the part was added via a 3rd party, long before I joined the project. As mentioned in https://github.com/fritzing/fritzing-parts/issues/252 , I plan to replace it. Also, with a recent pull request, we will add some more voltage regulators.

Hamsterjh, 2020-05-03:
Thanks for the feedback on this, much appreciated. Will work around it for the time being. Looking forward to future updates, great software.

KjellMorgenstern, 2020-08-19:
Wrong numbering of 7805 is tracked in #252 . Closing.

closed 2020-08-19 by KjellMorgenstern