fritzing/fritzing-app #3671: Cannot save files, posted error to page

1000FlowersBloom, 2020-05-03:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1 open fritzing
- Step 2 click save as

## Expected Behaviour
I can open fritzing and design a circuit but once I try to save it, my computer only tells me a vague error reading before crashing. 

[image]

KjellMorgenstern, 2020-05-04:
Can you reproduce this with the 348 build, or at least with the 0.9.4 release?

1000FlowersBloom, 2020-05-04:
Hello, yes this happens to me as well for the 0.9.4 release (the newest version if I'm correct), I purchased it yesterday.

KjellMorgenstern, 2020-05-04:
"saving files" is quite a basic and often used feature, and unlikely to be substantially broken. Without further details, there is not much we can do about this here.

1000FlowersBloom, 2020-05-04:
Are there any potential solutions that you can suggest for this issue? I've tried deleting and redownloading the program but that unfortunately still did not work. Any possible diagnosis on what seems to be the issue?

mMerlin, 2020-05-05:
Could your anti-virus program be blocking write access from Fritzing?

From the earlier post, this is failing when you click "save as", before you specify a file or location to save. Possibly before the file save requester is even displayed.

Something to try, just to get more information. Download one of the fzz sketch files, from a forum post (and probably rename it to get back to .fzz), open, make any change (even just select a part), then click save (not save as).  Does that work?

I would have suggested opening one of the examples, but modifying that, and clicking save opens the usual file requester anyway.  Which I was trying to avoid for this test.

Other ideas: is your system resource limited? Running out of memory at the wrong point can cause all sorts of strange errors. I would expect other programs to have problems too though. Even a web browser can use up a lot of resources if you open a few tabs.

vanepp, 2020-05-05:
I'd suggest deleting the user directories (which don't get touched during an install) in case they are corrupted. Here are the instructions on finding  and deleting them (which vary by operating system):

There are two user directories (with your parts and the parts database) which don't get touched during an install (to not affect your sketches during upgrades). On Windows they are in

c:\users\username\AppData\Fritzing\roaming\Fritzing (which is a hidden directory so you need to enable hidden directories in explorer) and

c:\Users\username\My Documents\Fritzing (where username is your windows id)

If you don't have any parts or sketches you want to keep you can just delete those two directories and Fritzing will recreate them, or you can move them aside by renaming them if you wan to keep something in them.

linux

~/Documents/Fritzing/parts
~/.config/Fritzing 

As well (if you haven't already). delete and reinstall the Fritzing app as well (although it sounds like you changed to 0.9.4 with no change, the user directories may fix that.) This sounds like something either in the code or the user directories is corrupted.

KjellMorgenstern, 2020-11-24:
Closing, can not reproduce the issue.

closed 2020-11-24 by KjellMorgenstern