fritzing/fritzing-app #3674: Successfully built but fail to install on Ubuntu 18.04.4

jiapei100, 2020-05-05:
## Current Behaviour
```console
Installing with make install...

========================= Installation results ===========================
make -f Makefile.Release install
make[1]: Entering directory '....../fritzing-app'
....../qmake -install qinstall -exe Fritzing /usr/bin/Fritzing
Error copying Fritzing to /usr/bin/Fritzing: Cannot create /usr/bin/Fritzing for output
Makefile.Release:57942: recipe for target 'install_target' failed
make[1]: *** [install_target] Error 3
make[1]: Leaving directory '....../fritzing-app'
Makefile:52: recipe for target 'release-install' failed
make: *** [release-install] Error 2

****  Installation failed. Aborting package creation.

Restoring overwritten files from backup... FAILED!

Cleaning up...OK

Bye.
```

**Build:**
<CD-548-16-g308f8d53 [Qt 5.14.2] >

**Operating System:**
<Ubuntu 18.04.4>

**Steps to reproduce:**
- `qmake`
- `make`
- `sudo make install`

## Expected Behaviour
Successfully install.

XayOn, 2022-05-16:
I'm pretty sure you're supposed to run that as root.

KjellMorgenstern, 2022-05-17:
Fritzing is usually installed and run in user mode. 
Installing Fritzing as root causes problems, especially with older versions, because some needed files can only be accessed as root during using Fritzing.

closed 2022-05-17 by KjellMorgenstern