fritzing/fritzing-app #3676: Open sketch from url (browser) fails to find some core parts

mMerlin, 2020-05-13:
## Current Behaviour

When using `Open with Fritzing` in a web browser (Firefox 76.0.1 (64-bit) in my case)) a fzz file link, Fritzing fails to find some core parts that do exist. Downloading the file, then opening that from Fritzing works. Fritzing also remembered (in recent) the downloaded and directly opened sketch file. Explicitly opening that also works.

[Forum post for example](https://forum.fritzing.org/t/can-i-export-my-project/9527/5)

**Build:**

Version 0.9.3 on some Linux
Version 0.9.4 (CD-498 2019-12-01) 64 [Qt 5.13.2] on Fedora 31
Currently unable to test with 0.9.5 ¦ Environment does not have Fritzing installed, so does not open from browser.

**Operating System:**

- unknown linux
- Fedora 31 5.6.8-200.fc31.x86_64

**Steps to reproduce:**

- Step 1 Click [Door Pi.fzz](https://forum.fritzing.org/uploads/short-url/uMlIle7HDQQBUHxKrnqUCVnZTt5.fzz) link from [Forum Post](https://forum.fritzing.org/t/can-i-export-my-project/9527/5) in web browser (Firefox 76.0.1 (64-bit))
- Step 2 Choose "Open with Fritzing", and click OK
- Step 3 From opens, downloads, then requestor with messages

```t
Unable to find the following 2 parts:

'Camdenboss_CTB0158-8_5_08mm_pitch_90deg_terminals'
at
'C:/Program Files/Fritzing/fritzing-parts/core/Camdenboss_CTB0158-8_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals'
at
'C:/Program Files/Fritzing/fritzing-parts/core/Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals.fzp'
```

- Step 4 Click OK on the message requestor
- Step 5 Document displays but without the terminal parts

## Expected Behaviour

Fritzing should open the sketch, and find the parts, when `Open with` is used from a web browser.

mMerlin, 2020-05-13:
There is some sort of interaction here with (probably) bad parts. Open Fritzing from the command line, search for "camdenboss" in parts. Shows 11 parts. Close Frtizing. Open fritzing from GUI now throws up requester with:

```t
Unable to find the following 11 parts:

'Camdenboss_CTB0158-3_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-3_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-11_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-11_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-7_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-7_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-12_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-12_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-4_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-4_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-2_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-5_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-5_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-8_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-8_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-10_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-10_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-6_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-6_5_08mm_pitch_90deg_terminals.fzp'
'Camdenboss_CTB0158-9_5_08mm_pitch_90deg_terminals'
at
'/home/phil/Documents/data_files/fritzing-parts/core/Camdenboss_CTB0158-9_5_08mm_pitch_90deg_terminals.fzp'
```

Close that requester, go back to parts search, and no parts are found. Close Fritzing, start from terminal, no requester, and the parts are back in the search results.

mMerlin, 2020-05-13:
Problem resolved. There were multiple part library repos on the system. The command line run used a script that pointed to one. The GUI startup used the one that came with the Fritzing flatpak app, Due to flatpak rules, and the way Fritzing was packaged, the library is a static snapshot old enough that the parts actually do not exist.

closed 2020-05-13 by mMerlin

KjellMorgenstern, 2020-05-13:
Thanks. One of the minor features on my list is a info dialog, that shows the location and some other stats about the database. I guess it would be helpful to identify such issues.

mMerlin, 2020-05-17:
That would also help for checking the current application state when switching libraries from the command line. When getting my process setup for checking changes to the core parts, while still using Fritzing for other things, I had some problems with Fritzing silently not doing what I wanted. Typos in the command or library path that ended up defaulting to 'base' library, with not enough (noticeable) information that there was a problem. Process has been tweaked, and mostly scripted, so does not happen any more.

Instead of a new dialog, some other applications include those details in the about dialog.