fritzing/fritzing-app #3677: GUI is blocked for up to 2 minutes until the check is complete

HeavyMoon, 2020-05-16:
## Problem
I built Fritzing for CentOS 7.
After startup, Fritzing will check for updates.
The GUI is blocked for up to 2 minutes until the check is complete.
The same problem occurs when I click "Check for Updates ..." in the menu.

## Proposed Solution
I want you to consider to Change to an asynchronous program or create an option to disable automatic update checking.

https://github.com/fritzing/fritzing-app/blob/308f8d531d251c0b4474cdf9b76f1dffb89dad49/src/version/versionchecker.cpp#L52-L63

KjellMorgenstern, 2020-05-16:
Indeed this should not block. However, usually this should also not take much time.

HeavyMoon, 2020-05-17:
Thank you for adding to the milestone!

closed 2021-07-25 by KjellMorgenstern