fritzing/fritzing-app #3679: Short-cut to draw new wire from connected pin with Prototyping Board

NilsRo, 2020-05-18:
## Problem
Breadboard view is made for breadboard. :-) If a wire is still connected it is not possible to draw a new connection from that point. You have to use an empty point as a workaround which is annoying for larger circuits using a Protoyping Board.

[image]

## Proposed Solution
Made an outline where not the wire is moved instead a new wire is drawn from that point as source are simple add a Key-combination which does not move wire instead it creates only new.

failiz, 2023-02-22:
This also happens in a normal breadboard. If there is a wire connected to a hole, you will select the end of the wire rather than creating a new wire.
A key press could make a new wire, but I am not sure if this is really necesary. In my opinion, holes in a breadboard should only have one cable attach to them. Of course, this has more sense in soldering protoboards, where you could have several wires soldered.

KjellMorgenstern, 2024-12-24:
I think this issue got less important with the release of Fritzing 1.0.4, which gives more precise and repeatable interactions when selecting items.
E.g. I was able to draw all these wires, using the point in the middle as starting point for each. I gets a bit tricky at the end, but you'll immediately see one of the wires highlight if the mouse pointer isn't on the hole. Also, the mouse pointer reliably destincts wether a new wire will be created, a bendpoint will be created, or an endpoint will be picked up.
[image]