fritzing/fritzing-app #3684: ubuntu bionic install script broken

mMerlin, 2020-05-31:
## Current Behaviour

The install_fritzing.sh script that comes with the version 0.9.4b download for (at least) bionic is broken. Several ways. I have a possible patch, but I do not know of a process to get changes into, or replacements for, the download files. That file name does not exist in the current source repository. Possible fixed script file attached. Renamed to append ".txt" to keep github happy. Works for me in the same environment (another fresh vm) as the reported failure.

[install_fritzing.sh.txt](https://github.com/fritzing/fritzing-app/files/4707168/install_fritzing.sh.txt)

```sh
% ./install_fritzing -u
 Copying ~/.mime.types -> ~/.mime.types.bak
cp: cannot stat '/home/phil/.mime.types': No such file or directory
Creating ~/.mime.types ...
Registering Fritzing MIME types...
Copying Fritzing data...
Entering fritzing/
Installing Fritzing MIME types...
-*-*-*-*-*-*-*-
xdg-mime: file 'icons/x-fritzing-fz.xml' does not exist
-*-*-*-*-*-*-*-

Setting up default application -> org.fritzing.Fritzing.desktop
-*-*-*-*-*-*-*-
-- TASK ENDED SUCCESSFULLY! --
-*-*-*-*-*-*-*-

Installing Fritzing system icons...
-*-*-*-*-*-*-*-
-- TASK ENDED SUCCESSFULLY! --
-*-*-*-*-*-*-*-

Making symlinks...
Doing final touch...
cp: cannot create regular file '/home/phil/.local/share/icons/hicolor/128x128/apps/fritzing.png': No such file or directory
grep: uninstall_fritzing.sh: No such file or directory
cp: cannot stat 'uninstall_fritzing.sh': No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
sed: can't read uninstall_fritzing.use: No such file or directory
Activating uninstall script...
sed: can't read uninstall_fritzing.use: No such file or directory
Activated!

cp: cannot stat 'uninstall_fritzing.use': No such file or directory
Updating databases...
Directory '/home/phil/.local/share/mime/packages' does not exist!
Done!
```

Examining the script file, and matching to the console messages found:

* wrong version string
  * `APPVER="0.9.3b"`
* for 'user' mode, it tries to backup the ~/.mime.types file without checking to see if it exists. It does check later if it exists, to create it if needed. Combine those to backup when it already exists, otherwise create it.
  * `cp "$MIMES" "${MIMES}.bak"`
* the `xdg-mime install` fails, because the first (as well as all of the other) xml file does not exist. a single icons/fritzing.xml file exists instead of the listed, per type, files
  * `xdg-mime install --mode "$MODE" 'icons/x-fritzing-«all».xml' && \`
* the code that is intended to detect and report problems with the xdg-mime installs does not report either success or failure for this failure mode (file does not exist)
  * `case $? in` … 0) … 1)
* several folders referenced by the (user mode) install may not exist. I tried to run on a bare ubuntu 18.04 vm, and got several complaints about target folders that did not exist during attempts to copy mime information and icon files. Again the failure was silently ignored. Some extra `mkdir -p` command lines are needed.
  * `xdg-icon-resource install --mode "$MODE" --context mimetypes --size 128 'icons/fz_icon128.png' application-x-fritzing-fz && \`
    * `cp: cannot create regular file '/home/phil/.local/share/icons/hicolor/128x128/apps/fritzing.png': No such file or directory`
* the sed command to adjust the icon in the desktop file does not do what the comment says it should. The comment says to keep the original, but the command backs up the original, modifies the original, then uses and deletes the backup.
  * `sed -i.use "s/icons\/fritzing_icon.png/${APP_ICON_ESC}/" org.fritzing.Fritzing.desktop`
* the sed command to adjust the icon in the desktop file does not work. The match string does not exist in the current desktop file.
  * `Icon=fritzing`
* the uninstall_fritzing.sh file does not exist. The whole block of code that is supposed to be modifying and activating it reports no such file or directory. Multiple times.
  * `DEVSTAT="$(grep 'DEVSTAT' uninstall_fritzing.sh | cut -d '=' -f 2 | tr -d '"')"`

**Build:**
Version 0.9.4
fritzing-a1ffcea08814801903b1a9515b18cf97067968ae-master-498.bionic.linux.AMD64.tar.bz2

**Operating System:**
Ubuntu 18.04

**Steps to reproduce:**
- create bare ubuntu 18.04 virtual machine
- download fritzing-a1ffcea08814801903b1a9515b18cf97067968ae-master-498.bionic.linux.AMD64.tar.bz2
- unpack
- cd to folder
- ./install_fritzing.sh

## Expected Behaviour

Either install properly, or report failure and exit from the install

KjellMorgenstern, 2020-05-31:
Yes, the script is broken. Fritzing must be run from the extracted archive, without running the script. The advise to do so (on the download page) is easily overlooked. I removed the script as a workaround, but this isn't a fix. Also, the script is still in the 0.9.4 release. 
The most severe scenario is that users run the script, it doesn't work as expected. Then they try to run it as root, which is even worse, as after that Fritzing can not be used properly as non-root user.
This is only mitigated because many users are quite tech savvy.
Further, the situation of MacOS and Windows is similar.

mMerlin, 2020-05-31:
Can you check/test my version of the script? See if it is sufficiently robust for more casual users.

I assume by "in the 0.9.4 release", you mean the download archive. Either that or the tagged commit in the repository. I hit it in the bionic download after following up from a forum post that started with issues getting access to the part library after installing, and turned into get installed properly.

Does running Fritzing from the archive folder get the mime type and other configurations done? Can fritzing sketches (or part files) be opened from the file browser? The smoke test after using my modified install script on a fresh 18.04 vm says it works there.

The fixes where not that difficult. A few hours, mostly making notes about what I was doing for reporting here, and verifying bash functionality. I work with CLI scripts a lot of places, but not regular enough to just do it. Need to lookup options in man pages, and test segments to see if they really do what I intended.

KjellMorgenstern, 2020-06-17:
There were some attempts made to comply with the freedesktop standard.
https://specifications.freedesktop.org/desktop-entry-spec/desktop-entry-spec-latest.html

We already have Ubuntu 18.04 (Unity?), would it make sense to test for more systems?
Like:

- [x] Ubuntu 18 + Unity
- [ ] Ubuntu 20 + Unity
- [ ] Ubuntu 20 + Budgie
- [ ] Fedora 32 + Gnome
- [ ] OpenSuse Leap 15.2 + Kde
- [ ] Mint 20 + Cinnamon

The script is more difficult to test than the average feature or bug:
1. it depends highly on the system and configuration 
2. it needs a fresh OS image each time for a reliable test
3. should also be backward compatible. Meaning, running the script again for a new Fritzing version should work as expected.

It seems to me that the tests would have to be manual (executed for each change to the script), but if there is a managed solution to automate such a test e.g. with a simulated input to a VM, i'd be interested...

mMerlin, 2020-06-17:
When I did the patch uploaded when the issue was created, I did not know that the script was known to be broken. Yes it needs to be tested against any environments it is expected to work for. And maybe some more as well, to make sure that it fails cleanly if the resources it expects to use are not available. The original was extracted from the bionic Fritzing 0.9.4 download, so the patch only **needs** to be compatible with where that runs. Having a version of the script that is forward, backward, cross os compatible is a bonus. All that is **needed**, is for the packaging process to include the correct version for the target environment.

I just downloaded the 3 different linux distro packages and verified that identical install_fritzing.sh files are included in each. So to (fully) promote my patched script does need some more testing. It has only been testing on a fresh build of Ubuntu 18.04 (based on ubuntu-18.04.3-desktop-amd64.iso)

The question about check/test in my previous comment was because I do now know what the current/historical procedure is for verifying install script functionality. From your comment, that amounts to running it on fresh/bare OS (fully updated) image install, with a smoke test to make sure it actually works. If you have any notes about specific things to test after install, I could use them here. I do not know of any tools to generically test install scripts. It would be possible, and not too difficult, to write a custom script (python?) to check the results of the install for a specific environment. That does not need to remain as generic.

When you say we already have Ubuntu 18.04, are you referring to my patched script?

With a little more information about the context(s) the script is expected to run in, I can do more testing for any environment that I can spin up a VM for. Currently that excludes Ubuntu 20. VirtualBox chokes on everything I have tried for that (full host system crash during guest OS install). I have not tried with Boxes or libvirt. I could add base Debian to that list of environments, which should be directly compatible with Ubuntu. I might be able to convince someone else to do a test against Arch linux.

To do anything around what I see looking over the freedesktop standard, I need a few details about what Fritzing is expected to support. For anything that could be different from the existing .desktop file. Fritzing has been translated to more than just [fi] and [ru], so additional entries should be added to the .desktop file, which needs a source for them. Currently only GenericName and Comment show translated entries, and only for the 2 languages.

If cleaning this up, beyond fixing what is actually broken, the install should ask the user for a few option choices. Like whether to use a manually installed parts library, the path to which could be put in the "Exec" entry in the desktop file. But that would be for after it actually works. Or it could be a new 'tool' script, to adjust the desktop file after installing.

I'll start by having a look at using it with Fedora 31/32. When I was doing the Ubuntu fixes, I thought I saw things that would need to be different for Fedora. I'll check closer.

mMerlin, 2020-06-19:
What is the current/initial scope for a fix for the install_fritzing.sh script? I have verified that it runs (user mode) successfully on fresh (VM) installs of Ubuntu 16.04, Ubuntu 18.04, Fedora 31, Fedora 32. As already mentioned, I can not currently test on Ubuntu 20.04.  Fritzing starts successfully in all of those environments from command line, file manager and gui. I can crash Fritzing in (at least) Fedora 32 with a simple smoke test. The issue is one of the wayland (and probably QT) compatibility issues.  I can work around the problem by setting environment variable QT_QPA_PLATFORM=xcb. That can be done in the desktop file, from the interactive command line, or more permanently/globally in the fritzing/Fritzing script file that is referenced from either interactive command line or gui runs.

So, the install script itself works, but Fritzing does not. For some environments. If we have a reliable way of detecting the problem environments (maybe XDG_SESSION_TYPE environment variable), the install script could be enhanced to compensate. Or add another install script command line argument to force the override only on user request (since the problem could go away after QT or Wayland updates). Similar for the Fritzing shared library dependencies. It is easy to detect problems (`ldd fritzing/lib/Fritzing | grep "not found"`), which could as a minimum be shown to the user during install, or the install script could (offer to) install the needed packages. Or again, add a command line argument to request the dependency installs.

Those extra options are beyond what the current broken script attempts to do. Should any of that be incorporated in the current patches, or should the basic functionality be fixed, and the other things pushed to another iteration? With notes added to the download page to help out people until then.

Where should the fixed file be pushed? The latest 0.9.5 development of course, but this also really should be put in a patch/fixes branch off of CD-498-0-a1ffcea, then used to create new 0.9.4 packages for download.

This current patch is not fully forward compatible. It still has the Fritzing version string hard-coded in the script. I do not know of a way to remove that. I did not find an obvious source for a current version string that the install script could use. Switching from wayland to x11 could be considered a forward compatibility item as well. Which might not be part of the install script. Directly in the fritzing/Fritzing script might be a better place to dynamically detect the running environment. For environments where the user could switch desktop environments, not all of which have wayland. Though forcing X11 in other environments is probably not a problem.

I have not tried system mode installs yet. If those remain 'broken', because of limited access to the parts library, should that installation option be removed? Until such time as it works properly? Potentially, the install script could set privileges on the fritzing-parts folder that would get it to work. Another change beyond the scope of what the current package scripts attempts, and something that would need to be a user choice. In class room environments, it is probably not wanted to allow regular users to be able to make changes to the library. So, `works properly` needs to be clarified.

KjellMorgenstern, 2020-06-19:
This is quite a complex issue, see also
https://github.com/fritzing/fritzing-app/pull/3290

I looked for a PR or commit,couldn't see your version of the script. Did you push it?

mMerlin, 2020-06-19:
My version is not in a repository anywhere right now. Just in my own working areas. When I first looked at it, I discovered that the file was not in the latest repo, and did not want to (at that time) add it in. I did not know that it used to be here, and had been deleted.

If you want a copy to test with on Ubuntu 20.04, I have not made any changes from the [install_fritzing.sh.txt](https://github.com/fritzing/fritzing-app/files/4707168/install_fritzing.sh.txt) uploaded in the start of this issue.  Original link just after the first paragraph.

This also overlaps the question above about where it SHOULD go.

The referenced PR is for a system only install. Some of the details have changed (and changed back) since that was done. I think the current script handles everything in that, a bit differently. But as mentioned, I have not tested system mode installs with it yet. User mode was all that the initial changes were for, in response to the problems reported in the forum.

mMerlin, 2020-06-19:
Using the bionic package, the install script worked fine on Debian 10 for user mode too.  Needed a reboot, but then the smoke test worked as well. Inspite of XDG_SESSION showing wayland. On cli start, I got a message `Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT-QPA_PLATFORM=wayland to run on Wayland anyway.`, so looks like Debian is configured to ignore wayland for this case.

rbrobots, 2020-09-18:
Thank you very much. The file you attached worked for me :)

gomker, 2021-01-06:
> My version is not in a repository anywhere right now. Just in my own working areas. When I first looked at it, I discovered that the file was not in the latest repo, and did not want to (at that [...]

I can confirm using the script I was able to install for `Ubuntu 20.04.1 LTS`
I did need to reinstall `libqt5core5a` but that may have been an issue just related to my workstation.

KjellMorgenstern, 2021-04-01:
I have replaced the previous script with a script based on mMerlins upload. Since the release 0.9.6, no issues where reported. I think this is solved for now. We plan to support more Linux systems soon, and will also test the script on those, but I think this issue can be closed. Feel free to reopen if you think that is not yet fully covered.

closed 2021-04-01 by KjellMorgenstern