fritzing/fritzing-app #3685: Ratnest wires out of the blue

s1Lema, 2020-06-01:
i was trying to do the circuit for a robot arm i'm making for school but then wires started appearing in the ratnest shorting everything, at first it hought it might be i wired incorrectly something but then i even tried to unplugg all the afected components but it still reported the phantom wires, if needed i should be able to send the screenshots of it unplugged, but hence i then have to reconnect everything i'd rather not do it unless needed.
Unfortunelly i don't know what made it start as by the time i realiced it it was happening all the time, but if needed i can send the file

[image]
[image]

**Operating System:**
<Windows 10>

deshipu, 2020-06-03:
In my experience this often happens when you have one of the resistors or capacitors reversed in one of the views compared to another view. Fritzing then takes both connections into consideration, and concludes that it's all a single connected network. Then it shows in your schematic view that the networks are not connected, while the other view clearly says they should, hence the ratnest line.

KjellMorgenstern, 2020-06-06:
@deshipu Thanks, that would actually be a nice opportunity to show a warning to the user, instead of silently getting the views misaligned.

failiz, 2023-02-22:
I think this is a very common problem and it would be nice to address it. My students find this problem when trying to use the simulator in several views which are misaligned. Change the title to make it more representative, such us: Warn users about misalignments between the views?

KjellMorgenstern, 2023-02-22:
I don't think that is possible to automatically decide wether this is accidental or intended, at least in the general case.

However, I don't think there are valid Fritzing circuits that connect both ends of a resistor or capacitor to the same subnet. Maybe those cases could be checked and warned?

failiz, 2023-02-22:
I think we should give a warning if they mismatch. The user should decide if it is intended or not. 
"However, there are probably no valid Fritzing circuits that connect both ends of a resistor or capacitor to the same subnet. Maybe those cases could be checked and warned?"
Sometimes that could be intended. For example, in pcb design, you could draw a trace that shorts a resistor. By default, you do not place the resistor but if you need to use it, you can cut the trace with a knife cutter.

Maybe we could improve the message that appears at the bottom "x of y nets routed". Something like that in the BB view:
"x of y SCH nets routed. z connections still to be routed. 
x of y PCB nets routed. z connections still to be routed. 
Warning: x connections in this view are not wired in SCH or PCB views."

failiz, 2023-02-22:
Look at this example. I have the following SCH:
[image]
And now I go to the BB view:
[image]
At this moment the BB view correctly says that there are 3 nets to route.

Now, I am going to shortcircuit a resistor with the blue wire and the output is in the next picture:
[image]

With my proposal, the text would be:
0 of 3 SCH nets routed.-  3 connections to be routed
Warning: there is one connection in this view that is not in the SCH view!
And it would be great if you could click on the warning and highlight that connection. 

In this example, I omitted the PCB view as has not been routed yet. If there were routes traced, the message would have to be extended to include routes from PCB and warnings that BB connections that are not in the PCB view.

KjellMorgenstern, 2024-10-01:
I created issue #4180 to plan something like a priority view to show this kind of information.
I think we can close this one in favor of #4180, since there are a number of issues with "ghost connections" that have been fixed since then.

closed 2024-12-04 by KjellMorgenstern