fritzing/fritzing-app #3686: Gerber export missing and moving some custom pcb cutouts

mMerlin, 2020-06-04:
## Current Behaviour

Exporting a pcb that uses a custom image with cutouts can leave out some of the cutouts, or put them in the wrong location. Using Version 0.9.4b. The Fritzing pcb view looks fine. Some reference information can be found in the [My custom pcb outline gerber export isn’t good… help!](https://forum.fritzing.org/t/my-custom-pcb-outline-gerber-export-isnt-good-help/9897/4) forum post. I did a search through the issues, and found 2 (both closed) that look like they could stem from the same cause.

#3015 Fritzing produces Gerber files with severe defects under Ubuntu 12.04 LTS x64
#3537 Gerber rendering moves and skips nodes in board outline

For the current case, I was able to get good exports by manual edits to the svg file used for the board outline. From the pattern of exclusions and missplaced cutouts, it appears that the gerber export is not handling path element relative move commands correctly. It appears (not confirmed) that each move ("m") is being positioned relative to the previous absolute ("M") command, disregarding the positions changes based on any interveneing relative move ("m") commands. So in a chain of "m" commands, only the first gets placed where intended. The rest are placed at the wrong coordinates. For cases where multiple identical cutouts are positioned in a array, with common offsets, they will will be placed on top of each other. For svg, that cancels (fills in) the previous cutout. So one cutout will show up, if an odd number of cutouts are specified with identical relative offset. None of them showup if there are an even number of copies. The fix that worked for the sample case, was to change all of the relative move ("m") offsets into absolute ("M") coordinates.

Tested using 0.9.5 as well. The export works there. Keep this around to document the issue for 0.9.4, and provide a potentially labour intensive work around.

**Build:**

Version 0.9.4.2019-12-01.CD-498 - Qt version 5.13.2
Version 0.9.5 (dCD-548-16-g6145d6d0 2020-04-19) 64 [Qt 5.12.7]

**Operating System:**

Fedora 31 workstation

**Steps to reproduce:**

## Expected Behaviour

Cutouts in the Gerber export should match the svg view

KjellMorgenstern, 2020-06-04:
I understand this is a duplicate of #3537 / #3015 and fixed on 0.9.5 ?

mMerlin, 2020-06-04:
The descriptions vary a bit. I have more specifics about the cause and a work around here. But yes. At least this specific case worked on 0.9.5. I did not try the examples from the previous issue, but the symptom description and images match what I would expect from the pattern/analysis here.

closed 2021-07-13 by KjellMorgenstern