fritzing/fritzing-app #3687: gerber export does not include final segment of shape created by "z" command in svg path element

mMerlin, 2020-06-04:
## Current Behaviour

Seen in Version 0.9.4b, but documented in 0.9.5.

A closed shape (or interior cutout) is normally defined in an svg path element using a series of connected straight and/or curved segments terminated by a close ("z") command. The close command indicates a straight line segment from the last active point back to the location of the previous move command. The gerber export is not including that final closing segment.

The gerber viewer screen shot is zoomed in to the upper right block of an export creating after loading the shown svg image (renamed to add .txt for github) for the pcb.

[noclosesegment.svg.txt](https://github.com/fritzing/fritzing-app/files/4726865/noclosesegment.svg.txt)
[image]

A much smaller/simpler svg would demonstrate this. I used what was already available from researching the issues from the [My custom pcb outline gerber export isn’t good… help!](https://forum.fritzing.org/t/my-custom-pcb-outline-gerber-export-isnt-good-help/9897/4) forum post.

**Build:**

Version 0.9.5 (dCD-548-16-g6145d6d0 2020-04-19) 64 [Qt 5.12.7]

**Operating System:**

Fedora 31 Workstation Edition

**Steps to reproduce:**

- Open a new sketch document
- Goto to pdb view
- select the default pcb part
- load image using included svg
- place a part of the pcb (to prevent warning during export)
- File » Export » for Production » Extended Gerber (RS-274X)…
- open the exported .gm1 file in a gerber viewer
- zoom in to the top right block, which for comparison is the first block in the svg file

## Expected Behaviour

SVG path element close commands should create a line segment in the gerber export.

KjellMorgenstern, 2020-06-04:
#3333 and more issues have been closed because we Fritzing got a new Gerber export, which than had to be rolled back. This 'z' closing issue seems not so difficult to fix, so should be done independent from introducing a new gerber export.

KjellMorgenstern, 2022-01-12:
The provided files demonstrates the same issue as in #3820

closed 2022-01-12 by KjellMorgenstern