fritzing/fritzing-app #3688: There is no fritzing.exe file in the .zip

Mareeo7, 2020-06-04:
Sorry But I didn't receive the fritzing.exe file in the .zip

KjellMorgenstern, 2020-06-04:
A fritzing.exe is in all our releases for Windows. Please check again. Also make sure to look at the extracted directory, not the archive. It can be easily overlooked within the archive, as many archive program rightfully don't highlight the file as executable as long as it is not yet extracted. Trying to execute the file without unpacking the zip file not work.
For further information, please see the instructions on https://fritzing.org/download

closed 2020-06-04 by KjellMorgenstern

turrithugger, 2020-06-04:
I don't think you understand. **It's not there.** I checked both the online file stored on the site and the extracted file, and there are no files labeled fritzing.exe. I even checked every folder manually and found no .exe files.

[image]

turrithugger, 2020-06-06:
@KjellMorgenstern are you still there?

Mouthpear, 2020-12-05:
I too downloaded the ZIP and there is no fritzing.exe file

Mouthpear, 2020-12-05:
> A fritzing.exe is in all our releases for Windows. Please check again. Also make sure to look at the extracted directory, not the archive. It can be easily overlooked within the archive, as many [...]

I too downloaded the ZIP and there is no fritzing.exe file

KjellMorgenstern, 2020-12-05:
Where did you download this zip from? 

I recommend that anyone without experience in C++ software development should use our official build from https://fritzing.org

Mouthpear, 2020-12-05:
> Where did you download this zip from?
> 
> I recommend that anyone without experience in C++ software development should use our official build from https://fritzing.org

From this very page.

KjellMorgenstern, 2020-12-05:
There are many downloads each day from https://fritzing.org/download , the zip files there of course have been checked carefully, and have not been modified recently. Actually, the download is paid, and I'd expect a lot more complains if the file was missing.

What makes you sure the .exe is missing?  Did you check the advise earlier in this issue?

iberotecno, 2022-05-11:
The .exe is not present. If you want to have it, yo may pay or install Visual Studio, all the dependences and compile it from source.  Is not easy but here you have the entire process. https://siytek.com/build-fritzing/

csmblair, 2022-12-28:
This is the most engineer thing I've seen. Why the hell wouldn't you include the .exe? I swear some engineers are literally robots.