fritzing/fritzing-app #3692: No .exe File

turrithugger, 2020-06-09:
This is a duplicate of #3688, which I think the developers skipped over way too fast. As evidence of this issue, I present:

- both windows and Github can't find it.
[image]
-A seven minute video of me looking through every file folder manually and finding no .exe files
https://www.dropbox.com/s/uyc7re0kzzre7db/iKaWxbvdN2.mp4?dl=0

KjellMorgenstern, 2020-06-09:
I think the issue might be that you expect the .exe file to be part of the source code?

aallan, 2020-06-13:
A .exe file is not part of the source code. There are periodic binary development releases, e.g. https://github.com/fritzing/fritzing-app/releases/tag/CD-548. Production binaries are distributed via the main Fritzing website, see https://fritzing.org/download/ for downloads for Linux, Mac, and Windows.

KjellMorgenstern, 2020-06-19:
Thanks aallan for explaining.

closed 2020-06-19 by KjellMorgenstern