fritzing/fritzing-app #3695: Pad Connection location not saving

master-puller, 2020-06-19:
## Current Behaviour
Fritzing does not save proper pad connections, and this almost made a project of mine fail completely

[image]

**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
Create any pcb with long, thin pads, and set the connection to one of the long parts.
- Step 2
Save and reload the file, taking note of where the traces lead.

## Expected Behaviour
Saves the correct pad connection, loads with correct connection on reopen

KjellMorgenstern, 2022-01-12:
I could reproduce the problem:
1. Set connection to north
2. Put a resistor north west of the pad and connect it
3. Resize the pad by grabbing it on the north west corner
-> Wire will jump between center and north.

When you release the corner, the wire will always jump to the correct position.
However, when loading the file, *and* this happens, there will be no release event
for the wire to jump to the correct location.

This is a big issue if you have many pads, as this will increase the likelihood of
the issue to happen.

KjellMorgenstern, 2023-05-19:
This is mostly solved. However, when the project is saved by pressing Ctrl-S while at the same time resizing the pad with the mouse, and not letting go the pad, the the wire connects to the center of the pad, not to the configured side.

closed 2023-07-18 by KjellMorgenstern