fritzing/fritzing-app #3697: Segmentation Fault when Deleting a part just after moving it with arrows

deshipu, 2020-07-02:
## Current Behaviour

Moving a part using the arrow keys in the PCB view, and then immediately deleting it using the del key results in a crash with a segmentation fault.

```
"arrow press event"
"move by arrow 0"
"delete item (1) 861851010 0 3 55ed1cd15a30"
"delete item (2) 861851010  3 55ed1cd15a30"
Segmentation fault
```

**Build:**
Version 0.9.5 (dCD-548-0-3d61c58 2020-02-07) 64 [Qt 5.9.5]

**Operating System:**
Linux

**Steps to reproduce:**
- Start a new project
- Switch to PCB view
- Drag a resistor onto the PCB and select it
- Move the resistor by pressing a down arrow on the keyboard
- Press del key to delete it
- Observe a crash

## Expected Behaviour
The part should be deleted and no crash should happen.

KjellMorgenstern, 2020-08-13:
Confirmed, fairly easy to reproduce.

failiz, 2021-03-18:
already released, close

closed 2021-03-18 by KjellMorgenstern