fritzing/fritzing-app #3699: Autorouter: places traces on both layers of single-sided PCB

hagen-git, 2020-07-25:
## Current Behaviour
Autorouter places traces on top layer of a PCB with 'one layer (single-sided)' layers property.

**Build:**
Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
<Windows 10> 64bit

**Steps to reproduce:**
1. Open PCB view
2. Click on PCB area
3. in Inspector pane, set 'layers' property to 'one layer (single-sided)'
4. Click on 'Autoroute'
5. Click on one of the yellow (top-layer) traces
6. Check Inspector pane Placement 'pcb layer' property

## Expected Behaviour
With PCB 'layers' property set to 'one layer (single-sided)', traces are only placed on one layer. (Usually the bottom layer, but maybe this could be configurable.)

## Workaround

### In PCB view

1. Select 'Both Layers' clickable
2. Autoroute
3. Menu-select: PCB > Routing > Select All Traces
4. Menu-select: PCB > Routing > Move to other side of board
    - now all traces are on the bottom layer
    - NB: you need to do this step twice, if you had 'Top Layer' clickable.

## Screenshots

### After 'Autorouter' invocation, PCB selected
[image]

### Trace selected that is on the top layer
[image]

### Also happens if only 'Bottom Layer' was clickable
[image]

KjellMorgenstern, 2020-08-13:
I used this arduino like circuit, set it to single layer and started the autorouter.
While the results of the autorouter varied, no traces where placed on the top layer. 

[Arduino-no-FTDI_single_layer.zip](https://github.com/fritzing/fritzing-app/files/5069330/Arduino-no-FTDI_single_layer.zip)

hagen-git, 2020-08-13:
[Here is a link to the design I made.](https://gitlab.com/repair1/conic-tg-621/-/blob/master/fritzing/video-amplifier.fzz) When using 'Autoroute', it routes top and bottom. If you then select the PCB, it should sit on 'single layer'. If you then select 'two layers' and again 'single layer', everything is on a single layer. 

Fritzing Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

Do you want me to install the latest development version and verify if the problem is still present?

hagen-git, 2020-08-13:
This is your design after loading and just clicking Autoroute with my Fritzing installation:
[image]

KjellMorgenstern, 2020-08-16:
There are two causes for Fritzing to fallback to 2 copper layers when loading the example https://gitlab.com/repair1/conic-tg-621/-/blob/master/fritzing/video-amplifier.fzz .

First (and fixed in PR 3705) is that the wires laid out on the breadboard use a copper1trace. This is not visible in the UI, only when examining the uncompressed fz file. I added a check to not use breadboard wires to determine the pcb layer count. This was probably the most common way issue 3699 is triggered.

Second, when loading the fzz, Wire 2552 is on copper1 in the pcb view. A single layer pcb does not have copper1. However, since the fallback only sets the UI to two-layer, but not the model, it is possible to add a copper1 wire to a single layer board. 
I did not fix this, as that would modify the fzz file only by loading it. A fix that would make sense to me would be to add a dialog, similar to the obsolete parts dialog, that offers to set the pcb layer count to two in the model.

failiz, 2021-03-18:
already released, close

closed 2021-03-18 by KjellMorgenstern