fritzing/fritzing-app #3701: Implement IPC Netlist Export to support AISLER Smart Test Infrastructure

Libermentix, 2020-07-28:
## Problem
Fritzing is a great tool for designing PCBs. Yet, it currently does not support the Export of IPC-netlist files. Hence AISLER Smart Test Infrastructure currently cannot be supported. 

See https://aisler.net/blog/electrical-testing-on-steroids

## Proposed Solution
Implement an export function that conforms to IPC Netlist standard (Similar to KiCad).

KjellMorgenstern, 2021-10-07:
Is there a preferred version? 
IPC-D-356A has been superseded by IPC-D-356B 
https://www.ipc.org/TOC/IPC-D-356B.pdf

Libermentix, 2021-10-07:
@KjellMorgenstern IPC-D-356B is preferred

KjellMorgenstern, 2023-07-14:
IPC export works with 1.0.0

This should be helpful for evaluating PCBs, both for automatic checks, but also for manual reviews or even 3rd party renderers: 
[image]

closed 2023-07-14 by KjellMorgenstern