fritzing/fritzing-app #3706: Please make development intentions clear

ChadBailey, 2020-08-18:
After a recent run-in letting the magic smoke out of some electronics I wondered to myself, "Can I simulate circuits so that I can test them before building them?" That is the question that lead me to Fritzing.

When I was on the website, I was asked for a $8 donation... (OK, 8 euro but I don't know how to do that sign and the actual amount is unimportant to my point). My first thought was, "Cool, an open source project which is requiring payment... I can get behind this!" My next thought was "Why are they asking for donations for development when they've only released 2 updates in the past 5 years?"

When I begin unearthing the layers searching, I find things like this https://github.com/fritzing/fritzing-app/issues/3435

Truly I'm left very confused and conflicted. Do I donate to help revive this project? Are the original owners just using the money donations on hookers as the aforementioned talk alludes to? If so, why not fork it? If not - maybe this is a new addition to the page to re-ignite dev work? If so, why has it been a year and a half since then? I have so many questions and no answers.

Truth be told, I would rather have spent $8 than the time and effort I've spent trying to figure out if I should toss a coin in the hat so to speak. I know that if I feel this way - very likely most visitors feel this way. Worse yet, I'm sure they wouldn't bother investigating and will just download the latest release on github foregoing the donation option. I don't blame them for doing so, but I for some reason decided to dig in and attempt understanding.

TLDR; if you want donations, make it super clear that they are to rekindle dev work. If the donations are actually being used for that, make it clear on the page that the project is under new management/in process of being revived. If they aren't being used for that, please get over your nostalgia of the name and fork the project already. Whatever you do, don't call it "libre-fritzing" or "free-fritzing" because those names suck.

KjellMorgenstern, 2020-08-18:
We consider #3435 solved and closed, I think it should answer those questions already.  However, the complete ticket is quite tedious to read and understand, so I try to summarize it here.

Fritzing development had ceased for three years, until I became maintainer.
I understand you want to know about the usage of the payments. I believe the best way to answer this is to deliver a perfect and up to date software. Unfortunately, we inherited quite a backlog of technical and non-technical issues. Development is slower than I wish it were, but with the CI and the new website, I think we are on the right track and increasing speed. Ups, sounds like one of those manager reports from my former employer :-) 

The problem of Fritzing was that there were simply no developers anymore, also due to a lack of money. The payments go to the development of Fritzing. Although many still consider it a donation due to the open-source nature, the payments are not donations; tax is paid.
A lot has happened since I became maintainer and primary developer. To name a few things, only listing technical aspects, not legal or business:
 Repair the builds, setup a continuous integration, update for more recent platforms (Window 10, MacOS Mojave, Ubuntu, Fedora), several hundred github issues, several thousand support emails, one new release so far, several emergency fixes to the website, finally migrating the 2010 website implementation to a 2020 software stack, a new forum server, moving the blog from wordpress to a static page, a dedicated email server (increasing email delivery from 95% to more than 99%) and automated testing for the website. Because of the website rewrite, there was low activity on fritzing app repo the last months.
In progress are 
* release 0.9.5 
* a new parts editor
* improvements to the project editor on the website
* adding more parts

On the website, right above the payment form, we state "... This way, we can ensure future releases, bugfixes, and features". We could elaborate this more, and I did in the forum, in issue #3435 and others. On the webpage, people don't need to bother with the project's struggles if they don't want to. I'll prepare a similar story for the Fritzing Blog, and will mention this on the main website.

Does this answer your questions?

KjellMorgenstern, 2020-08-18:
PS: Fritzing can help you avoiding magic smoke.  But it is *not* a simulator, by most definitions of a simulator. This is the first point in our FAQ. Fritzing can detect if you connect Vin to GND by highlighting the connections. But it won't warn you if the 4Ohm resistor between 12V and GND does not have sufficient cooling ;-) or try to simulate any of those Arduino or Esp boards variations.

WimberleyTech, 2020-08-18:
Chad,
    
    Thank you so very much for your post.  It pissed me off so much that
    I am going to make another direct donation to Kjell.  This will be
    my third.
    
    In the many many years I have lived on this earth, I have learned
    that one can place people into two categories: givers and takers.
    
    I have only posted on this group a couple of times, but I watch the
    thread EVERY DAY.  
    
    Kjell, thank you for the good work you are doing and also the others
    that I think are helping in the background.  Keep up the good work.
    
    Please confirm that your paypal account has not changed since my Oct
    23 transaction.  
    
    doug
    
    
    
    On 8/17/2020 11:07 PM, Chad Bailey
      wrote:
    
    
      
      After a recent run-in letting the magic smoke out of some
        electronics I wondered to myself, "Can I simulate circuits so
        that I can test them before building them?" That is the question
        that lead me to Fritzing.
      When I was on the website, I was asked for a $8 donation...
        (OK, 8 euro but I don't know how to do that sign and the actual
        amount is unimportant to my point). My first thought was, "Cool,
        an open source project which is requiring payment... I can get
        behind this!" My next thought was "Why are they asking for
        donations for development when they've only released 2 updates
        in the past 5 years?"
      When I begin unearthing the layers searching, I find things
        like this #3435
      Truly I'm left very confused and conflicted. Do I donate to
        help revive this project? Are the original owners just using the
        money donations on hookers as the aforementioned talk alludes
        to? If so, why not fork it? If not - maybe this is a new
        addition to the page to re-ignite dev work? If so, why has it
        been a year and a half since then? I have so many questions and
        no answers.
      Truth be told, I would rather have spent $8 than the time and
        effort I've spent trying to figure out if I should toss a coin
        in the hat so to speak. I know that if I feel this way - very
        likely most visitors feel this way. Worse yet, I'm sure they
        wouldn't bother investigating and will just download the latest
        release on github foregoing the donation option. I don't blame
        them for doing so, but I for some reason decided to dig in and
        attempt understanding.
      TLDR; if you want donations, make it super clear that they are
        to rekindle dev work. If the donations are actually being used
        for that, make it clear on the page that the project is under
        new management/in process of being revived. If they aren't being
        used for that, please get over your nostalgia of the name and
        fork the project already. Whatever you do, don't call it
        "libre-fritzing" or "free-fritzing" because those names suck.
      —
        thread.
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3706",
"url": "https://github.com/fritzing/fritzing-app/issues/3706",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

KjellMorgenstern, 2020-08-18:
@WimberleyTech Thanks for your support, my account didn't change.

ChadBailey, 2020-08-18:
@KjellMorgenstern forgive me, I posted a bug report asking for clarity yet I myself was not clear.

Thanks for summarizing all of that for me, super helpful. I will outline what I think are clear steps to fulfill what I'm asking.

Regarding using as a circuit simulator - just because that question lead me here doesn't necessarily mean I expect fritzing to be able to deliver that functionality. This does tie in to an important point though. Questions like this should be easily answered by the website particularly if the user can't simply download the program to see if it does the thing they hoped.

## Requesting
- Make a new blog post talking about development process
- Post new blogs regularly to keep future visitors from thinking the reboot of development failed (about every 90 days shouldn't be unreasonable to ask)
- Update the sentence referencing the price
  > We are asking you to pay 8€ (less than US$9) for downloading the application. This way we can ensure future releases, bugfixes and features.
  - This is confusing because "asking" implies it's a donation or optional. Suggested verbiage:
  > Fritzing is open source, but costs $8 for a personal license and $25 for a business license. Click [here] for more information.
  - The [here] link should link to a page or even just blog post outlining what you get with either license - and at your option an explanation of why.

## Benefit Hypothesis
- Much more likely to capture new users
- New users = more contributors ultimately

WimberleyTech, 2020-08-18:
You've sent €15.39 EUR to kjell
    Spend this on your favorite Hefeweizen, or something else that
      brings you joy!!
    
    
    
    
    On 8/18/2020 1:06 PM, Kjell wrote:
    
    
      
      @WimberleyTech
        Thanks for your support, my account didn't change.
      —
      [
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/fritzing/fritzing-app/issues/3706#issuecomment-675631207",
"url": "https://github.com/fritzing/fritzing-app/issues/3706#issuecomment-675631207",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

el-j, 2020-08-26:
Hi, Long Time No "read".

First of all thank you Kjell for all this work done. I had you position a while and I know how much work the drifting universe is...

Anyways I needed a drifting version for osx Catalina and I was a bit shocked that I can download fritzing for free from computer bild but not from fritzing.org. 

I am not sure if a "force" to pay is the right way to go...but I understand it.

Besides that I build a GitHub action release flow on my fork that builds the current master for osx Catalina and pushes it to a private nextcloud Share. The DMG is running, and I will do some tests with it the next days as I need it for a project myself.

So thanks again for all your work. If you are interested in migrating the builds to GitHub actions instead of Travis let me know. 

So FYI: this master build is running on osx Catalina 15.6 ;)

d3cline, 2020-10-15:
My 2 cents, charging for GPL software is kinda sketch. I am going to go learn KiCAD. This kinda bums me out since I released a lot of free and open source content with fritzing because I expected it to remain free. Now that content is behind  a pay wall. Trying to move software like this to a paid structure just feels wrong, even if I understand the altruistic motives. glhf

KjellMorgenstern, 2020-10-15:
Well, the community is an amazing, and huge. I want to take this oportunity to thank everyone who is supporting the project, with their payment or even with their work. Fritzing is alive and kicking ass, thanks to you!

@d3cline Few people can afford to work for free over multiple years. But I don't think that means we have to scrap the project, including all the content that was created for it. Obviously I don't follow this inverse "rather dead than red" idea.
 I am curious, what kind of content did you provide? Did it really not help with your business or paid job in any way?

ChadBailey, 2020-10-16:
> My 2 cents, charging for GPL software is kinda sketch. I am going to go learn KiCAD. This kinda bums me out since I released a lot of free and open source content with fritzing because I expected it to remain free. Now that content is behind a pay wall. Trying to move software like this to a paid structure just feels wrong, even if I understand the altruistic motives. Maybe the community should come together or let it die, move to a stronger platform etc. Darwin! glhf

This is the _real_ problem behind putting the software behind a paywall. First, it's not really fair to anyone who contributed prior to it being put behind a paywall. Next, it causes unnecessary strain to downstream dependent projects. If it had been that way up front, I don't think there would be a problem.

At the end of the day, it really seems kinda clear that if no one has the time or money just let the project do whatever it's going to do... fall into obsolescence, hand over to someone else to maintain, be forked, be rewritten from the ground up in QBASIC by a random benevolent stranger... this is the way of all software right?

ChadBailey, 2020-10-16:
My last post was a little more focused on the merit so making a second regarding the original bug report purpose.

Just want to say that while the advice I gave wasn't followed exactly, I do see that you've made effort to improve to this end and thanks for that.

WimberleyTech, 2020-10-16:
I grew up in the "Age of Aquarius."  (telling my age)  What happened?  8 bucks or euros...whatever.  Hell I have been donating directly to Kjell just because I feel like it is the right thing to do.  You can download the source, n'cest pas?  Anyway, I am happy to see progress on the tool and looking forward to a new release.  Peace!

failiz, 2020-11-17:
Should we close this issue? It is not a bug. The discussion can continue in the forum if anyone needs to add something else.

hkienle, 2021-02-03:
> I have been donating directly to Kjell

How can I do that?

KjellMorgenstern, 2021-02-03:
The simplest way is via https://fritzing.org/download

I operate the site, and since a few weeks also the company "Fritzing GmbH" behind it. All the previous organizations have been integrated or merged into this one company. Things have been cleaned up a lot, and i am getting to put more effort into the application itself again.

hkienle, 2021-02-03:
Thanks, great to hear!