fritzing/fritzing-app #3707: Version 0.9.3 is broken from ubuntu repos.

abr4xas, 2020-08-19:
## Current Behaviour
With a fresh install:

[image]
[image]
[image]
[image]
[image]
[image]
<please attach screenshot if possible>
**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.9.1]

**Operating System:**
Ubuntu 18.04.5 LTS

**Steps to reproduce:**
- apt install fritzing

## Expected Behaviour
work without problems or errors (?)

KjellMorgenstern, 2020-08-19:
This looks like an issue with the Ubuntu package of 0.9.3, which is not maintained here.

abr4xas, 2020-08-20:
Hey @KjellMorgenstern thanks.

closed 2020-08-20 by abr4xas