fritzing/fritzing-app #3711: Improve placement of PCB components like Protoshield; Protoshield was mirrored because default alignment was bottom.

cof-celldeg, 2020-08-27:
I just blew an evening of routing a pcb layout by not realizing that a protoshield component was apparently on the "bottom" layer, thus had its connectors mirrored. I never changed the alignment. I was using the Protoshield PCB as a template to fit my components in; at that time this sounded like a killer feature which made me chose fritzing over KiCAD. Now the layout cannot fit anymore and I will have to reroute.

Which brings my to my complaint:

PCB components should _not_ be able to have a pcb layer assigned. Or at the VERY least they should by default be assigned to the top layer. 

I consider this important because: 

Anything else is highly unintuitive. I noticed more or less in time (when I was checking the orientation of an IC before soldering) and didn't waste many physical things. But this took some amount of knowledge and having been bitten by this in earlier projects. I assume this to be very hard to notice for a novice which kind of is your target audience.

KjellMorgenstern, 2020-08-27:
Could you add more instructions on how to reproduce the problem? Which protoshield did you use, which components did you add? 
While some designs intentionally only use one side, it is very common to place components on both sides of a PCB, so this is an important feature to have. 
Connector names are shown when you hover the mouse over them. Did you use that feature? Or is this not related to the issue you describe? Did you use the breadboard or schematic view?

cof-celldeg, 2020-08-28:
I could upload the project, but I suspect it will not help much with any forensic analysis. What I did was roughly to edit a schematic that included the "Arduino Prototyping Shield" at V4.0 (shipped with a default installation). Then I switched into PCB mode and started routing. 

I tried to recreate the issue, and noticed that the bottom placement is triggered when "view from below" is active (opposed to "view from above", even if the component is added in the schematic mode. 

I cannot remember switching to this either, but would not be surprised if I did by accident; It is close to the task bar and is more easily misclicked. If so, I did not notice during placement.

Regarding top vs. bottom placements, I agree with the requirement to have placement on both layers for general components. It is just an issue that if the component is a PCB itself, it makes the whole affair very unintuitive to use, because it changes the meaning of top and bottom for the placed components.
If I wanted components to be on the bottom of the Protoboard, i would likely have switched the components placement instead. On the other hand, with the stacking behaviour of shields, I can kind of see the use case of flipping a whole PCB, to put it above another component's pin headers for example. So this seems to be a requirement too.

After thinking through all of this, I reduce my call to have the schematic editor disregard settings regarding "view from above" and "view from below". It is only a schematic with no notion of top and bottom. I assume default placement for anything I drop there.

I assume it would be more intuitive if PCB components disregard this in general. If I drop a PCB into the PCB view, and I have "view from below", I would assume that its placement is correct and I now see the bottom layer. This is not what currently happens.

KjellMorgenstern, 2020-10-08:
I think you are not the first who run into this, and maybe there is an easy fix.

So this only applies to PCB components, like arduino shields. Placing such a component on the bottom layer means placing it upside down. This is very likely not what the user wanted. 
In such a case, it a warning could be shown to the user, suggesting to place the PCB component on the top layer, like
------
"You are placing a PCB component with "edit bottom layer" active, this would revert the orientation of that component. We recommend to always place PCB components at the top layer."

[Ok, place at top]          [I want it upside down]
-----

Would that have prevented it?

cof-celldeg, 2020-10-09:
If that's possible that would most certainly help. If it is possible to detect that a component includes a PCB, this should occor for every such component. 

I would suggest following message:

The component you are inserting contains a printed circuit board (PCB). Because the option "edit bottom layer" is active, the PCB's front layer will be placed in your design's bottom layer. This means that the component will be placed upside-down, which is not recommended. "Change to top layer" disables the option for this placement only.

[Change to top layer] [Continue editing bottom layer]

There is a second aspect: The option should invariably be disregarded when placing new components from the schematic editor. Maybe add a second setting, with the same effect, but visible as a checkbox within the schematic editor. The biggest issue I had is that the setting is only visible in the layout window , but controls actions performed in the schematic editor (inserting component into schematic regards the edit bottom layer setting)

KjellMorgenstern, 2020-10-09:
Sounds good to me, also the usage of top layer when inserting from other views than PCB. 

Detecting wether parts have a PCB, not only holes, copper or silkscreen, should be possible.