fritzing/fritzing-app #3715: Independent rotation for parts and labels in schematic

supuflounder, 2020-09-03:
## Current Behaviour
For reasons that escape me, the resistors in my schematic (thru-hole) come out on my breadboard either reading left-to-right or right-to-left.  So, I thought that if I rotate the resistor in the schematic so the pin relationships are changed, the color bands will all work out correctly.  That seems to work.  But it also puts the labels upside-down, which makes no sense at all.  I should be able to rotate the part or its label independently.
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
Text

KjellMorgenstern, 2020-10-08:
Please add version information and reproduction steps. The bug template is there for a reason. Screenshots might help too in this case.

KjellMorgenstern, 2020-10-08:
I added 10 resistors in a schematic. All resistors showed up in the same orientation in breadboard view. Without further information, this issue should be closed.

failiz, 2021-11-15:
Not enough info. The labels rotation is being discussed in #684

KjellMorgenstern, 2022-01-21:
Closing as duplicate of #684

closed 2022-01-21 by KjellMorgenstern