fritzing/fritzing-app #3718: parts moved in one window move in another

kelbycriswell, 2020-09-16:
## Current Behaviour
I have spent a bit of time working with the breadboard window and when i finally get everything where i want it. I go to the Schematic or PCB view to get those looking nice. When I return back to the breadboard view i find that the parts are all moved around and no where near where i placed them. Is there a way to unlink the parts from one view to another so that they can be positioned independently of each other.  
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.3 (b5c895d32 2016-04-19) Cocoa [Qt 5.12.2]

**Operating System:**
<Windows 10>

KjellMorgenstern, 2020-10-08:
That is odd, aligning parts in one view does usually not affect the other. 
Can you add more information, like a file to start, and the instructions on which part to move in wich view?
Also, please add the version information.

failiz, 2021-11-15:
No answer from the reporter and not enough information to reproduce it. Close until we get more info?

closed 2021-11-15 by KjellMorgenstern