fritzing/fritzing-app #3720: IC pin labels lost on any change to part properties. Undo doesn't work.

marcin-jozwikowski, 2020-09-20:
## Current Behaviour
When a generic IC has custom-names pins they get lost on any change of that part properties. 
I've tested it with properties such as _Hole Size_; _Pin Spacing_; _Number of Pins_ and _Package Type_.
That wouldn't be so much of a problem if I could revert that with Ctrl+Z but it reverts only the change made but doesn't bring back the labels.

[image]

After changing _Hole Size_ to _Large_ all labels are gone

[image]

**Build:**
Version 0.9.5 (dCD-548-0-3d61c58 2020-02-07) 64 [Qt 5.9.5]

**Operating System:**
Ubuntu 20.04.1 LTS

**Steps to reproduce:**
- Add an IC part
- Edit pin labels
- Make any change to the part like _Pin Spacing_

## Expected Behaviour
Most of all I'd expect Undo to work properly.
I'd also like for the labels to stay untouched on any change when pin count stays the same. 
When lowering the count (from 32 to 16 for example) label over the pin count could be deleted but it would be better for them to stay as well. That sort of change could be accidental.

KjellMorgenstern, 2021-11-22:
Fix for Undo is in the pipeline. 
Fix to remember pin labels after certain operations will be implemented as issue #2336

KjellMorgenstern, 2023-06-23:
Just retested to be sure it works now. Pin labels now persist when other properties change.
Even changing the number of pins works, unless you reduce it, then the labels for pins that have been removed will not persist, even if you increase the number again.

closed 2023-06-23 by KjellMorgenstern