fritzing/fritzing-app #3721: Is it possible to make an auto router to do straight lines, not curved?

KhArtNJava, 2020-09-22:
Is it possible to make an auto router to do straight lines, not curved?
Before auto routing:
[image]
After auto routing:
[image]
What I expected to see (I'm used to straight lines):
[image]

<please attach screenshot if possible>
**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Open Attached file [1.zip](https://github.com/fritzing/fritzing-app/files/5263021/1.zip)
- Open Schematic
- Press Autoroute

## Expected Behaviour
What I expected to see (I'm used to straight lines):
[image]

KhArtNJava, 2020-09-22:
[Fritzing.zip](https://github.com/fritzing/fritzing-app/files/5263055/Fritzing.zip) Fritzing dir from my documents folder, with models, that I used in this sketch.

KjellMorgenstern, 2020-10-08:
I think the offset you observed is a bug. I could reproduce it with your file.

KhArtNJava, 2022-01-06:
Hello. Any news on this bug? :)