fritzing/fritzing-app #3722: Auto interface feature

pronoym99, 2020-09-29:
### Add an auto interface feature for the breadboard view

- The pin names and their annotations should be displayed for every part/component on a window
- Two components can be selected at a time
- **Concerned pins would be automatically connected** OR **the user shall specify for each pin on component 1, the pin to which it is to be interfaced to on component 2**
- Can be extended later for simultaneous interfacing of multiple (more than two) components

KjellMorgenstern, 2020-10-08:
That looks to be a narrow use case. In fact, I don't really see an advantage. What rule should be used to identify "concerned" pins? No matter what, the user must still decide for each individual connection if it is correct. That might even become harder if connections are automatically made based on an rule. The pure process of connecting pins with a mouse is very fast, making sure that it is correct and routing the connection is hard.

pronoym99, 2020-10-08:
Well I don't believe that the use case is narrow given the fact that _concerned pins_ are most of the times GPIOs of controller boards. Even if that is not the case, listing pin names of two devices alongside each other gives us an additional advantage - it helps us remember which pins have been interfaced with each other especially if multiple ICs/boards enter a design. 

For e.g. when interfacing an IR sensor with an Arduino Uno board, the 3 pins of the IR sensor must appear on the left (in the proposed feature). Then the user can directly select one of the power pins of the Uno board from the right and similarly for ground pins. The only pin remaining is the digital input pin which will have a drop-down list of the digital pins on the Arduino Uno board (on the right) to be interfaced with. As soon as this is done, the user presses the **Auto-interface** button and voila !!!. 

Although seemingly similar to the auto-route feature in Eagle, implementation is easier, this being a part of the breadboard view.