fritzing/fritzing-app #3724: Question regarding branches not meged into develop

failiz, 2020-10-09:
Hi @KjellMorgenstern ,
I can see that there are a lot of branches that have not been merged into the develop branch. For example, these are branches that fix issues and I do not see why they should not be merged: 3683, 3699, 3338, italian, 2186
These are branches that implement some functionality and maybe you are still working on them or maybe they are not useful any more: svg_optimize, refactor_projectstructure, copperfill, enable_cpp14, qt5.9, legacy_builds, obsolete_scripts, QVector, win32-zlib-fixes.  But maybe you forgot about them...
Or am I missing something?
Cheers.

KjellMorgenstern, 2020-10-09:
I'd prefer to discuss this on the individual #PRs and not open a generic issue.

failiz, 2020-10-11:
Sorry for the noise. I double check the code and I can see that a lot of them have been merged correctly. Apparently, Github is using "Squash and merge" instead of "pull request merges". This makes that SourceTree does not show them as merged into development.

closed 2020-10-11 by failiz