fritzing/fritzing-app #3726: Option to import image file to *.fzz project

KhArtNJava, 2020-10-18:
## Problem
Sometimes I need to look at pinout image of some elements.
Now I need to switch to another app (iame viewer) to look at it. But it will be cool, if there will be an option to paste image to Fritzing Schematic from MS Word (via copy/paste) or to import *.png or *.jpg files and display them to Schematic.
Image block could be like Note (Edit - Add Note) block - resizable, movable with mouse drag and drop.

## Proposed Solution
To add image import to Fritzing.

KjellMorgenstern, 2020-10-18:
Shouldn't that pinout be the same as in the PCB view?

Parts have the option for a (basic) markup documentation. Images are not supported, but URLs . Would that help as a workaround?

Could you provide an example part?

KhArtNJava, 2020-10-19:
Hello.
I mean, some pins of elements (for example, esp8266) can be configured for different roles.
ESP8266 model that I use has only one marking of each port (pin) - D0, D1, etc:
[image]

But, for example, D0 port could be used as WAKE:
[image]

So, I want to paste image from MS Word or LibreOffice Writer to Schematic, to see all possible port usage.
It could be like this (I drew it in Paint.Net):
[image]
 
It could be not only image with pionut. It could be any image, for example, some photo or logo...

HaraldRau, 2020-11-12:
Is it a solution to describe the different roles in the metadata? They are displayed when "mouse over".

KjellMorgenstern, 2020-11-12:
You can add a special part that allows to load an image. However, the schematic view will only allow monochromatic images.

[image]

[image]

The breadboard view supports color images.