fritzing/fritzing-app #3734: user ic (movable quad and dual opamp) parts loose ratsnest connections when moved to different board sides .

magstripp-r, 2020-10-22:
## user ic (movable quad and dual opamp) parts loose routing connections when moved to different board sides .

<please attach screenshot if possible>
**Build:**
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
<Windows 8.1>

**Steps to reproduce:**
- Step 1: go to pcb layout mode
- Step 2: flip pcb side for component 

## Expected Behaviour
connections should be intact.
[new_trap10.zip](https://github.com/fritzing/fritzing-app/files/5425347/new_trap10.zip)

vanepp, 2021-02-27:
This isn't a Fritzing bug, but rather that the TL074 part you used is incorrectly configured. Here is a corrected part (also posted to the forum post where you got the original part if that is easier) that should correct the problem. The Tl074 part is (among other problems) missing the copper1 definition in all the pcb layer sections fo the .fzp file. To use the part below you will first need to select all the tl074 parts in your sketch and do a delete minus on them. That deletes the part, but leaves the traces connected to it. You then need to delete the TL074 part in your mine parts bin and then shutdown and restart Fritzing to really clear the original part. Then down load the part below, and remove the trailing .zip (this is really a .fzpz file which is zipped but github won't load it without the trailing .zip.) You can then use fiile->open in Fritzing to load the new part and drag that in to your sketch. You then need to grab the end of each of the connections and drag it over the associated pin of the new part to make the connection (it should change from red to green when connected.)

[TL074-Quad-Opamp-(Movable)-improved.fzpz.zip](https://github.com/fritzing/fritzing-app/files/6053661/TL074-Quad-Opamp-.Movable.-improved.fzpz.zip)

You are usually better off to ask about things like this in the Fritzing forums, as you see I rarely remember to read this ...

Peter

failiz, 2022-03-29:
Not an error. Close

vanepp, 2023-11-16:
> The improved part doesn't seem to fix this. Anyway, Fritzing should handle this in some way.

My bad! I obviously didn't check the original complaint after fixing the errors in the part. You are correct it still breaks. Moving the part from bottom to top deletes the rats nest lines (which it shouldn't be!) Interestingly, undo will not undo that change as well which I think it should. Hitting undo back to the start does not restore the rats nest lines that were deleted. More problems too. I just tried to do a delete minus in schematic (because I couldn't figure out in pcb which chip the TL074 was and it deleted only the subpart I selected not the entire chip as expected. 

[image]

Do you want to open separate issues on the various issues found or leave them all here (as it is likely all connected to subparts?) If you like I can poke at this failure and figure out all the ways it can fail. It is possible that the bottom of the board rats nest error is a more general issue, or it may be related to only subpart parts. It is fairly unusual to have parts on the bottom of a THT board so there may be a more general issue there or it may be related to only subpart parts. Testing is the only way to find out I expect.  Well the new part loaded all on its own reacts as it should:

[image]

switching it to the bottom of the board reacts as I would expect (this is with my improved part though, it is possible the original part is causing internal corruption somewhere!) 

[image]

The rats nest lines are still there and connect as expected. I'll continue poking at this and see if I can figure out what is going on.

closed 2024-06-26 by KjellMorgenstern