fritzing/fritzing-app #3739: [FEATURE] Dark theme

PinkDev1, 2020-10-29:
Right now, there are pretty much no theming options on Fritzing, and I'd be really nice if it had them; Prolongued use of fritzing with that #000000 background gets tiresome pretty fast

KjellMorgenstern, 2020-10-29:
Nice to have, but currently unrealistic given the number of open issues.

closed 2020-10-29 by KjellMorgenstern

Bluscream, 2022-05-29:
Still unrealistic?

MimiVRC, 2024-04-29:
not having theming options seems to be breaking the app in general now. many things are black deck on a dark gray background because there is no way to force it to just be light mode

KjellMorgenstern, 2024-05-02:
Colors in Fritzing are determined in many places:
- SVGs from the parts (thousands of files, many more 3rd party files)
- User settings (only very few yet)
- OS settings (themes, dark mode)
- Qt defaults
- Hardcoded within the Code
- Within QSS stylesheets, which can be distinct, specific to OS and Window Manager
- Within generated styles during runtime

It is both, 1. a difficult styling problem. Because of the huge number of elements that need to be restyled, and because just inverting or applying a palette does not work.
2. a big technical problem due to the many different patterns that styles are applied, and not all Operating Systems react similar.