fritzing/fritzing-app #3741: item not removed from the scene before deleting it

failiz, 2020-11-01:
## Current Code
the exportAux function in mainwindow_export.cpp adds a watermark to the scene, but the watermark is not removed from the scene before deleting it.
First, we add the watermark to the scene (the watermark is added in addWatermark function):
`QGraphicsItem * watermark = m_currentGraphicsView->addWatermark(":resources/images/watermark_fritzing_outline.svg");`

Then, we render the scene and after the watermark is deleted:
```
if (watermark) {
		delete watermark;
}
```

## Expected Code
```
if (watermark) {
                m_currentGraphicsView->scene()->removeItem(watermark)    
		delete watermark;
}
```
 Or is this not necessary?

KjellMorgenstern, 2020-11-01:
From the Qt Documentation

> `void QGraphicsScene::addItem(QGraphicsItem *item)`
> Adds or moves the item and all its childen to this scene. This scene takes ownership of the item.
> 
So yes, the scene takes ownership. Also, removeItem would take back the ownership and we would have to care about deleting it. But, from the destructor of QGraphicsItem:

> `[virtual] QGraphicsItem::~QGraphicsItem()`
> Destroys the QGraphicsItem and all its children. If this item is currently associated with a scene, the item will be removed from the scene before it is deleted.
> Note: It is more efficient to remove the item from the QGraphicsScene before destroying the item.

So, it is ok if we just delete it, but your suggestion is more efficient, according to the Qt docs. I don't think efficiency is relevant (it is just a single item). But more important: We are not leaving a dangling pointer in the scene, both ways are correct.

failiz, 2020-11-01:
Thanks for the explanation. I agree that it is not necessary to change the code. Closing the issue.

closed 2020-11-01 by failiz