fritzing/fritzing-app #3746: PCB subparts

KjellMorgenstern, 2020-11-11:
## Problem
When using Arduinos to creat an Arduino shield, it could be useful to have ICSP headers and mounting holes by default, but be able to remove them later on, in case they are not needed.

## Proposed Solution
Add a mechanism similar to schematic submodules, but for the pcb view.

Alternative / Workaround: Duplicate the parts in question for variants with and without the headers. Line out drill holes in the silkscreen, and user can then easily manually add drill holes where required.

Are there more use cases then just the Arduino Mega/Uno sized boards?

mMerlin, 2020-11-12:
This would be able to be applied to any of the breakout boards / modules as parts, where the board/part has mounting holes that may or may not want to be applied to the pcb it is going to be attached to. It would change the general recommendation of leaving mounting holes out of parts, to include them as sub-parts, so that they can be removed per project / per instance.

This type of submodule probably needs to be on/off only. Moving them relative to connectors seems like a good way to not have things work.

An extension would be to have multiple alternate sub modules. Tor example, to provide an option to have different sized holes, without creating completely separate part variants.

failiz, 2021-01-13:
This is related to #3775 (how to handle multiple properties without having to create parts with all the variants).