fritzing/fritzing-app #3747: Fritzing 0.9.4 crashes on MacOS Catalina 10.15.7

afiskon, 2020-11-11:
## Current Behaviour

```
Process:               Fritzing [76494]
Path:                  /Applications/Fritzing.app/Contents/MacOS/Fritzing
Identifier:            org.fritzing.Fritzing
Version:               0.9.4 (bda4e)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Fritzing [76494]
User ID:               501

Date/Time:             2020-11-11 14:20:33.055 +0300
OS Version:            Mac OS X 10.15.7 (19H2)
Report Version:        12
Bridge OS Version:     4.6 (17P6610)
Anonymous UUID:        CEF6EE2A-84E4-E420-CD3A-8BBA3AA98AE2

Sleep/Wake UUID:       0F65A0E5-C9B8-4601-8EDF-4C217F8D7D39

Time Awake Since Boot: 690000 seconds
Time Since Wake:       3100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000020
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [76494]

VM Regions Near 0x20:
--> 
    __TEXT                 000000010a52b000-000000010ad8d000 [ 8584K] r-x/rwx SM=COW  /Applications/Fritzing.app/Contents/MacOS/Fritzing

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.fritzing.Fritzing         	0x000000010a89ca4a ModelPart::modelPartShared() + 10
1   org.fritzing.Fritzing         	0x000000010a750d65 ItemBase::setUpImage(ModelPart*, LayerAttributes&) + 37
2   org.fritzing.Fritzing         	0x000000010a5f2484 SvgIconWidget::setupImage(bool, ViewLayer::ViewID) + 164
3   org.fritzing.Fritzing         	0x000000010a5f2220 SvgIconWidget::SvgIconWidget(ModelPart*, ViewLayer::ViewID, ItemBase*, bool) + 496
4   org.fritzing.Fritzing         	0x000000010a5ef485 PartsBinIconView::setItemAux(ModelPart*, int) + 213
5   org.fritzing.Fritzing         	0x000000010a5ea8af PartsBinView::addPart(ModelPart*, int) + 15
6   org.fritzing.Fritzing         	0x000000010a5eeca5 PartsBinIconView::addPart(ModelPart*, int) + 21
7   org.fritzing.Fritzing         	0x000000010a5e578c PartsBinPaletteWidget::addPart(ModelPart*, int) + 76
8   org.fritzing.Fritzing         	0x000000010a5fd4d7 BinManager::search(QString const&) + 567
9   org.fritzing.Fritzing         	0x000000010a5e83e2 PartsBinPaletteWidget::search() + 98
10  org.qt-project.QtCore         	0x000000010bee7713 QMetaObject::activate(QObject*, int, int, void**) + 1843
11  org.qt-project.QtCore         	0x000000010bee7713 QMetaObject::activate(QObject*, int, int, void**) + 1843
12  org.qt-project.QtWidgets      	0x000000010b121ec8 QWidgetLineControl::processKeyEvent(QKeyEvent*) + 552
13  org.qt-project.QtWidgets      	0x000000010b118bc2 QLineEdit::keyPressEvent(QKeyEvent*) + 34
14  org.qt-project.QtWidgets      	0x000000010b016979 QWidget::event(QEvent*) + 521
15  org.qt-project.QtWidgets      	0x000000010b117f25 QLineEdit::event(QEvent*) + 725
16  org.qt-project.QtWidgets      	0x000000010afdbe80 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 272
17  org.qt-project.QtWidgets      	0x000000010afdd401 QApplication::notify(QObject*, QEvent*) + 1041
18  org.fritzing.Fritzing         	0x000000010a55fa1a FApplication::notify(QObject*, QEvent*) + 26
19  org.qt-project.QtCore         	0x000000010beb7714 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
20  org.qt-project.QtWidgets      	0x000000010b034926 0x10afcc000 + 428326
21  org.qt-project.QtWidgets      	0x000000010afdbe80 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 272
22  org.qt-project.QtWidgets      	0x000000010afdd235 QApplication::notify(QObject*, QEvent*) + 581
23  org.fritzing.Fritzing         	0x000000010a55fa1a FApplication::notify(QObject*, QEvent*) + 26
24  org.qt-project.QtCore         	0x000000010beb7714 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
25  org.qt-project.QtGui          	0x000000010b5accb7 QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) + 167
26  org.qt-project.QtGui          	0x000000010b59338b QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 219
27  org.qt-project.QtGui          	0x000000010b58dcb5 QWindowSystemInterface::flushWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 581
28  libqcocoa.dylib               	0x000000010df185ea 0x10deeb000 + 185834
29  libqcocoa.dylib               	0x000000010df187c8 0x10deeb000 + 186312
[... 453 lines cut ...]
USB Device: Touch Bar Display
USB Device: Apple Internal Keyboard / Trackpad
USB Device: Headset
USB Device: Ambient Light Sensor
USB Device: FaceTime HD Camera (Built-in)
USB Device: Apple T2 Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 55.3
Thunderbolt Bus: MacBook Pro, Apple Inc., 55.3
```

**Build:**

Build 0.9.4 for Mac OS X Mojave, release_downloads/307 (the latest one available for MacOS)

**Operating System:**

MacOS Catalina 10.15.7

**Steps to reproduce:**

Look for various parts for a while, type something like "FT50-43", "SMA", "Antenna" in the search box, at least this is what I did. After 10-20 attempts Fritzing will crash.

## Expected Behaviour

The software doesn't crash.

failiz, 2020-11-12:
I can reproduce this on Windows in 0.9.4. It seems related to the number of items that appear in the search. It does not crash with "Antenna" or "SMA", but if I search for "we", it crashes after 4 or 5 searches.

KjellMorgenstern, 2020-11-12:
Confirmed with dev branch in debug mode. Did not happen on first search for "we". But searching for it a second time resulted in a nullptr call

KjellMorgenstern, 2020-11-12:
Quick note, had to shift to differnt tasks: to just keep it from crashing, the dirty way. 
``` svgiconwidget.cpp
LayerAttributes layerAttributes;
	m_itemBase->initLayerAttributes(layerAttributes, viewID, ViewLayer::Icon, ViewLayer::NewTop, false, false);
	ModelPart * modelPart = m_itemBase->modelPart();
	FSvgRenderer * renderer = nullptr;
       // avoids the segfault , blank rectangle will be shown as icon
	if (modelPart) {
			renderer = m_itemBase->setUpImage(modelPart, layerAttributes);
	} else {                  
                        DebugDialog::debug(QString("error icon %1").arg(m_itemBase->m_filename));
			DebugDialog::debug(QString("error icon %1").arg(m_itemBase->m_id));
        }
```

KjellMorgenstern, 2021-03-18:
Hm... actually I never found the root cause for this. The fix just detects the situation, and a blank rectangle instead of the expected icon is shown.
Offload this to the sqlite database, and remove the caching mechanism? A search in sqlite should be way faster than what we currently have, and less error prone.

closed 2021-03-18 by KjellMorgenstern

failiz, 2021-03-18:
> Offload this to the sqlite database, and remove the caching mechanism? A search in sqlite should be way faster than what we currently have, and less error prone (that would mean also storing the icons in sqlite)

Let´s open a new bug report for this then.