fritzing/fritzing-app #3748: Text size of the connection pins of the ESP32 too large - text overlap

Gekla, 2020-11-11:
## Current Behaviour
Text
<please attach screenshot if possible>
**Build:**
<e.g. Version 0.9.4 (CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
<Windows 10>

Hello fritzing team,
I am a teacher at a vocational school and we have various projects that we implement with the ESP32 microcontroller. I imported the ESP32 into the component list using the "ESP32-38PinWide.fzpz" file using the import function. However, the font size of the pin labeling of the ESP in the breadboard view is too large and text overlaps. The connection designations are therefore hardly legible.
[image]

Interestingly, the font size is displayed correctly in the accompanying SVG file and there are no overlaps. 
[image]

Even after installing the Fritzing fonts, the error persisted.
Can you give me a hint as to what is causing this error and how to fix it?
regards
Gekla

HaraldRau, 2020-11-11:
Hallo,

kannst Du mal das Paket "ESP32-38PinWide.fzpz" hier einfügen. Dann sehe ich mir es an und versuche zu helfen.

Gruß Harald!

HaraldRau, 2020-11-11:
Hallo noch mal,

welche Schriftarten hast Du installiert und mit welchen Programm hast Du das Teil bearbeitet oder angezeigt?

HaraldRau, 2020-11-12:
Um das Paket "ESP32-38PinWide.fzpz" hier einzufügen, musst Du den Dateinamen in "ESP32-38PinWide.fzpz.zip" ändern.

HaraldRau, 2020-11-12:
The font is arial, not OCRA. I will change the font.

HaraldRau, 2020-11-12:
I've converted the font into paths. Certainly not the best solution, but it works for now.

[ESP32-38_Pin_V_01.fzpz.zip](https://github.com/fritzing/fritzing-app/files/5532055/ESP32-38_Pin_V_01.fzpz.zip)

Download the file and remove the extension *.zip

KjellMorgenstern, 2020-11-14:
Instead of Arial, why not use OpenSans?

HaraldRau, 2020-11-14:
I can't say, but "arial" was used in the svg-file - whatever reason. I changed the font to OCRA. But I still exported the font in paths. So the Part is working.

KjellMorgenstern, 2020-11-14:
The recommendation is to use OCR-A or OpenSans, Fritzing includes those fonts so the text don't need to be vectorized in paths or embedded to the svg files. Don't use "px" values in svg documents (the font-size attribute uses them here). Since svg are scaled with the viewBox, the px value would loose its meaning, specifically, it will not correspond to a screen pixel. 

The pcb view of the part seems to have the same issue.

HaraldRau, 2020-11-15:
O.K., I see I have to learn more about the part editor and the Fritzing part file system.
Is this documentation up to date?
[https://fritzing.org/learning/tutorials/creating-custom-parts](https://fritzing.org/learning/tutorials/creating-custom-parts)
Can the part be used as a reference?
[https://fritzing.org/media/uploads/learning/graphic_standards/fontsandtemplates.zip](https://fritzing.org/media/uploads/learning/graphic_standards/fontsandtemplates.zip)
Is there another part that can be used as a reference?

HaraldRau, 2020-11-18:
Is it possible to move this issue to fritzing-parts?

KjellMorgenstern, 2020-11-18:
It's a request about 3rd party part, not a core part, and there is no response yet from @Gekla . Should be closed.

KjellMorgenstern, 2020-11-24:
Closing due to inactivty

closed 2020-11-24 by KjellMorgenstern