fritzing/fritzing-app #3765: A part with female connectors do not connect to a part with male connectors when moving it.

failiz, 2020-12-04:
A part with female connectors do not connect to a part with male connectors when moving it. 

Version 0.9.5 in development
Windows 10

**Steps to reproduce:**
- Go to breadboard view
- Add a resistor outside the breadboard
- Move the breadboard and place it under the resistor, so the resistor´s connectors are touching the breadboard´s holes

The resistor is not connected to the breadboard.

[image]

## Expected Behaviour
The resistor should be connected to the breadboard.

## Context
This came up as I was trying to make a part (a probe for a multimeter) that could be attached to male and female connectors. As there are only male or female types, I made a part with a male and a female connector at the same place which internally connected through a bus. Then I realized that this only works when the part with the male connector is moved to the part with the female connector.

KjellMorgenstern, 2020-12-08:
Not sure if there are any use cases where this unilateral connecting makes sense. So, I think it's a bug. The parts should be connectable both ways.

vanepp, 2021-03-01:
"Not sure if there are any use cases where this unilateral connecting makes sense. So, I think it's a bug. The parts should be connectable both ways."

There is in fact a purpose to the current connection scheme. In parts, a male connector connects to a female connector and visa versa. A wire connects to either sex. To connect a male part pin to a male part pin requires a wire between them. The reason is because breadboard pins are female (and most part pins should be male.) So the part connects to the breadboard. The place where female is used is on connectors (such as the ISCP connectors on the Arduino) that are at 90 degrees to the other connectors and if placed on the breadboard would short (because of the vertical short in the breadboard strip.) Those connectors are set to female and thus won't connect to the breadboard (and short!), but can be connected to with a wire. There is also a pad designation (in parts editor), which acts female (it won't connect to the breadboard) and is likely related to SMD parts, but I don't know function it has. 

Peter

KjellMorgenstern, 2021-03-01:
@vanepp With "connectable both ways" I was meaning those ways: 1. place the breadboard, add the resistor, it is connected
2. place the resistor, only then add the breadboard. They do not connect, but they maybe should?

Actually, I would not like to apply this behavior to the breadboard, as there is a risk that it starts to pick up parts to eagerly.

But for the case with the multimeter it makes sense. Another solution could be to create a wire with a femal endpoint, so if you directly want to connect the multimeter to a pin, you would switch the wire to a different one that has a socket.

failiz, 2021-03-01:
@KjellMorgenstern "Actually, I would not like to apply this behavior to the breadboard, as there is a risk that it starts to pick up parts to eagerly." If we do not add the breadboard in the new behaviour, we fix the issue with the probes of the multimeter but we still have the other issue: We allow that two components that should be connected together (based on its position and one connector touching the connector of the other part) are not connected. As in the case of the image that I reported. This seems weird to me. (And excluding the breadboard would make the code less clean).
What would be the scenario where the new behaviours is dangerous?

vanepp, 2021-03-10:
"2. place the resistor, only then add the breadboard. They do not connect, but they maybe should?"

I think this is part of the "delete minus" reconnect the existing traces api. To make the connection you need to select and move the part (or the wire in the case of delete minus):

Two resistors without the breadboard:

[image]

drag in the breadboard and the resistors don't connect:

[image]

click on the resistor and move it one slot left then back and it connects (but moving the breadboard does not cause the connect, possibly because of female pins?)

[image]

mgesteiro, 2021-05-23:
This "bug" was "hit" also in the forums here

https://forum.fritzing.org/t/28byj-48-stepper-motor-and-uln2003-driver-module/12883/8

when connecting two parts together: so there are more use cases out there.