fritzing/fritzing-app #3767: Remove unnecessary null pointer checks

elfring, 2020-12-08:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [PartsBinPaletteWidget](https://github.com/fritzing/fritzing-app/blob/df2ed3d70e77e55d867e4e6c008c7731ef082595/src/partsbinpalette/partsbinpalettewidget.cpp#L141 "Destructor for PartsBinPaletteWidget")
- [ItemDrag::cleanup](https://github.com/fritzing/fritzing-app/blob/df2ed3d70e77e55d867e4e6c008c7731ef082595/src/itemdrag.cpp#L39 "ItemDrag::cleanup function")
- [SvgIconWidget::cleanup](https://github.com/fritzing/fritzing-app/blob/df2ed3d70e77e55d867e4e6c008c7731ef082595/src/partsbinpalette/svgiconwidget.cpp#L121/sdts/sdtsindexedreader.cpp#L85 "SvgIconWidget::cleanup function")

KjellMorgenstern, 2020-12-08:
Indeed, it is save to call delete null_ptr . 
Also PartsBinPalette might set m_icon to nullptr after deleting.
Are you working on a PR?

elfring, 2020-12-08:
How do you think about to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability-delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

KjellMorgenstern, 2020-12-08:
Of course, use it. I do. But still, fixes should not break anything. This is problematic without automated tests: Reviewing a fix for a harmless warning takes often more time than fixing an actual real bug that had been reported on github.