fritzing/fritzing-app #3768: Improve exception safety with smart pointers

elfring, 2020-12-08:
Would you like to wrap any pointer data members with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

Update candidates:
- [GraphUtils::chooseRatsnestGraph](https://github.com/fritzing/fritzing-app/blob/54715fba164709c55e5f1515dd5659f8147091e5/src/utils/graphutils.cpp#L354)
- [Grid](https://github.com/fritzing/fritzing-app/blob/df2ed3d70e77e55d867e4e6c008c7731ef082595/src/autoroute/mazerouter/mazerouter.cpp#L321)

KjellMorgenstern, 2020-12-08:
I prefer not... new / modified code may use newer styles. You may also apply it if you suspect a bug do to unclear instance ownership / livespan. But refactoring thousands of lines of code only because there is a different method, just gets in the way of tracing bugs in the history, possibly even causing bugs.

KjellMorgenstern, 2020-12-08:
Please also note: Qt is not using exceptions, in general the framework is not (yet) exception save. 
Maybe interesting approach to find issues is to try to find places in Fritzing that are relying on throwing exceptions.

elfring, 2020-12-08:
The mentioned source code places contain risks according to questionable software behaviour if an exception like [`std::bad_alloc`](https://en.cppreference.com/w/cpp/memory/new/bad_alloc "Documentation") would actually be thrown.

KjellMorgenstern, 2020-12-08:
If you want to harden Fritzing against out of memory errors, then i think there are other techniques needed. Preallocating sufficient memory during startup, for example. That's useful for embedded systems, but not so much for desktop applications?

elfring, 2020-12-08:
Would you like to take [the C++ guideline “R.11: Avoid calling new and delete explicitly”](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "Recommendations around allocation and deallocation") into account?

closed 2021-02-27 by KjellMorgenstern