fritzing/fritzing-app #3771: Export as image, toggled layers not honoured

alex-code, 2020-12-20:
## Current Behaviour

Toggling visibility of layers and using `Export->as Image->PNG` or `Export->as Image->JPG`  can show wire artefacts or not export selected layers.

**Example Stripboard**
[image]
**After toggling the wire layer visibility**
[image]

I've also had times where unselecting the wires or parts layers, they still appear in the exported imaged.

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
Windows 10

**Steps to reproduce:**
- Create a stripboard sketch
- Toggle layers
- Export as image

## Expected Behaviour
No wire artefacts and only selected layers exported as part of image.

failiz, 2020-12-21:
I can reproduce this in 0.9.4. Steps to reproduce that I use:
Go to breadboard view and and a wire with a couple of angles over the holes of the breadboard
Toggle the wire layer off
Export as an image (this image is correct)
Toggle the wires layer on 
Export as an image (this image has the artefacts and the wire is missing).

I checked in the latest development version available and this issue is still reproducible.
As soon as yo move the position of the wire, then it exports correctly.

failiz, 2021-03-18:
already released, close

closed 2021-03-18 by KjellMorgenstern