fritzing/fritzing-app #3777: load/save a project

Loris58, 2021-01-16:
when I load a project, at 90% it is already visible then at 92% the application closes unexpectedly and it is not possible to open it from Fritzing. To open it, I have to click on the project directly before opening Fritzing and having the project launch Fritzing.
This is strange because it doesn't happen with all projects. Some times it happens also when perform save with name.
## Current Behaviour
Text

**Build:**
Version 0.9.4 
(CD-498-0-a1ffcea 2019-12-01) 64 [Qt 5.12.3]

**Operating System:**
<Windows 10 pro

KjellMorgenstern, 2021-01-16:
By "<Windows 10 pro", do you mean you have a version lower than Windows 10?

Maybe you have to different instances of Fritzing on you computer. One which is associated with the project, and started when you click on the project. And a different one, when you start Fritzing directly, and then open the project in Fritzing.

KjellMorgenstern, 2021-02-27:
There is not enough information to work on this issue, or the information has become unavailable. Therefore we close this issue. If you still experience this issue, please provide as much information as possible. Some minimal information could be 1. Version of Fritzing . 2 Your operating system, including the version and architecture (eg. 32 bit, ARM, ...) 3. How often does this happen? 4. A screenshot showing the issue, 5. Files that cause the issue.

closed 2021-02-27 by KjellMorgenstern