fritzing/fritzing-app #3778: Linking Errors while Building

babbott1000, 2021-01-18:
## Current Behaviour

Linking Errors

[image]
[image]

**Build:**
- b8c2430b09123770bffe079d756643d78957c1a5
- Qt 5.15.2, Qt Creator 4.14.0
- MSVC 2019, 64 bit
- Microsoft Visual C++ Compiler 16.7.30517.126 (amd64)
- CMake 3.19.2

**Operating System:**
- Windows 10 (Build 19041.746)

**Steps to reproduce:**
- Build with Qt 5.15.2 and MSVC 2019 64-bit

## Expected Behaviour
Successful Build and Execution

babbott1000, 2021-01-19:
I'm unsure if this is related, but there are several errors in `libgit2detect.pri` when I attempt to build fritzing

[image]

JPTa, 2021-08-19:
Still same issue with latest sources of frizing, same tools used as above. Somehow I think this is not gonna be top priority for anyone to fix as it would not help getting some money.

BrunoLindorfer, 2021-09-24:
According to [this issue](https://github.com/fritzing/fritzing-app/issues/3801#issuecomment-808774104) `BUILD_SHARED_LIBS` should be activated.

KjellMorgenstern, 2021-09-24:
There are many ways to get errors when building a project like Fritzing. 
Most fixes we receive for building Fritzing are clean and valid in some way, but specific to an individual developers' setup, and can break for other developers. It would cost us significant resources to test them, all while in our setups it builds just fine, and we have hundreds of real open bugs. Also, many of the build fixes may work on one system, but the build then behaves erratic or crashes on other users systems. And that even without the conflict of interest JPTa mentions.

closed 2021-09-24 by KjellMorgenstern