fritzing/fritzing-app #3779: Arduino pro mini se se cale pas sur la grille de la strip board

Claude-Pierre, 2021-01-22:
## Current Behaviour
Les pins principales de Arduino Pro mini ne se verrouillent pas sur sur la grille de la strip board.
Seules les 6 pins de programmation sont verrouillées (BLK GND VCC TX0 RX1  GRN °
<please attach screenshot if possible>
**Build:**
0.9.4

**Operating System:**
<Windows 7
[image]
>

**Steps to reproduce:**
- Step 1
- Step 2

## Expected Behaviour
Text

KjellMorgenstern, 2021-01-26:
Usually, the the header PINs should automatically align. If they don't, you can disable "align to grid" in the "View" menu.

closed 2021-01-26 by KjellMorgenstern

Claude-Pierre, 2021-01-26:
I tanks you for you answer..

CHOULET-PARPILLON Claude