fritzing/fritzing-app #3782: Arduino Nano Export SVG Missing

calum904, 2021-01-26:
## Current Behaviour
When exporting a sketch as SVG which includes the Arduino Nano, the Arduino Nano is missing from output SVG. PNG works fine.
[image]

**Build:**
Version 0.9.3 (b5c895d327c44a3114e5fcc9d8260daf0cbb52806 2016-04-19) 64 [Qt 5.12.5]

**Operating System:**
<Linux/Ubuntu>

**Steps to reproduce:**
- Step 1Create Sketch
- Step 2 Add Arduino Nano
- Step 3 Export to SVG

## Expected Behaviour
Arduino Nano to appear in output

## Fix
As like similar exports, the ID for the Arduino nano (found /usr/share/fritzing/parts/svg/core/breadboard/Arduino Nano3_breadboard.svg) contains **breadboardbreadboard** where it should be **breadboard**. Manually removing the extra **breadboard** fixes the problem.

KjellMorgenstern, 2021-01-26:
@calum904 Thanks for the proposed fix. 
Related to #1252

failiz, 2021-03-10:
@calum904 We would really appreciate a pull request to fix this. You can even do it through the website without installing any tools. The link below explains how to do it: 
https://docs.github.com/en/github/managing-files-in-a-repository/editing-files-in-another-users-repository
And that file in the part repository is here: 
https://github.com/fritzing/fritzing-parts/blob/develop/svg/core/breadboard/Arduino%20Nano3_breadboard.svg

KjellMorgenstern, 2021-03-10:
After applying the fix, will a previously created project load fine, and now correctly export to svg?

If we can make sure that there is no regression for existing projects, i would replace all the occurrences in the parts repository. Otherwise, we need to use the obsoleting / replacement mechanism.

vanepp, 2021-03-13:
"After applying the fix, will a previously created project load fine, and now correctly export to svg?"

Yes it should work fine. While I'm no where near an expert on obsoleting parts, I believe obsoleting is only necessary if the change causes a physical change in position of a drawing element in the svgs. This change is only to the layerId which won't change any physical positions and the part should load fine on older versions. Running FritzingCheckPart.py in directory mode against the core parts repository will flag all the parts with missing layerIds and then someone would need to look at each part to see if the layerId is in the svg but doesn't match the layerId in the associated .fzp file (which sounds like the nano case) or if the layerId is entirely missing which would need a either a new group added to the svg or as I just discovered works, the id of the starting svg group to be changed to the layerId (which FritzingCheckPart currently complains about even though Fritzing accepts the layerId fine.)

closed 2021-07-25 by KjellMorgenstern