fritzing/fritzing-app #3784: Compilation Failure on OS X Big Sur

samuelboland, 2021-02-04:
## Current Behaviour
Attempting to compile following instructions here: https://github.com/fritzing/fritzing-app/wiki/1.-Building-Fritzing leads to a failure on OS X Big Sur. 

[image]

Error text copied here: 

> :-1: error: no such file or directory: '/usr/lib/libz.dylib'
> :-1: error: no such file or directory: '/System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation'
> :-1: error: no such file or directory: '/System/Library/Frameworks/Carbon.framework/Carbon'
> :-1: error: no such file or directory: '/System/Library/Frameworks/IOKit.framework/Versions/A/IOKit'
> :-1: error: no such file or directory: '/System/Library/Frameworks/Security.framework/Versions/A/Security'
> clang: error: no such file or directory: '/System/Library/Frameworks/Security.framework/Versions/A/Security'
> make[1]: *** [../release64/Fritzing.app/Contents/MacOS/Fritzing] Error 1
> make: *** [release] Error 2

[**Build:**]
Latest git version on 'develop' branch with SHA b8c2430b09123770bffe079d756643d78957c1a5

**Operating System:**
OS X 11.2 (Big Sur)

**Steps to reproduce:**
- Step 1
Attempt to compile following instructions. Get as far as clicking Run in QT Creator.

## Expected Behaviour
Compilation proceeds successfully. 

## Notes
I believe that this may due to a similar issue to this Stackoverflow question: https://stackoverflow.com/questions/62587131/macos-big-sur-python-ctypes-find-library-does-not-find-libraries-ssl-corefou

It appears that Apple has removed some libraries from the filesystem and instead stores them in some sort of cache. This is not my area of expertise so I am unable to comment further.

KjellMorgenstern, 2021-02-27:
I think you can let qt handle finding the library instead of the current way to specify the path. This might break building on older versions however.
Are you attempting a universal binary build?

prescindivel, 2021-09-28:
any solution?

thenickdude, 2022-01-03:
In `/phoenix.pro` I had to edit it to replace those old Framework file references (that were broken with Big Sur's move to a combined dyld cache). These were the original lines:

```bash
    LIBS += /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    LIBS += /System/Library/Frameworks/Carbon.framework/Carbon
    LIBS += /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
```

Which I replaced by:

```bash
    LIBS += -framework CoreFoundation
    LIBS += -framework Carbon
    LIBS += -framework IOKit
```

I also used these to link to my MacPorts versions of libraries:

```bash
    LIBS += /opt/local/lib/libssh2.dylib
    LIBS += /opt/local/lib/libiconv.dylib
    LIBS += /opt/local/lib/libz.dylib
```

Also in `/pri/libgit2detect.pri` I had to change this line:

```bash
            LIBS += $$LIBGIT2LIB/libgit2.a /System/Library/Frameworks/Security.framework/Versions/A/Security
```

To:

```bash
            LIBS += $$LIBGIT2LIB/libgit2.a -framework Security
```

thecodeguyexamples, 2023-01-30:
I followed the steps above but I am still getting a runtime error.

> :-1: error: framework not found libz.dylib

What did you replace `LIBS += /usr/lib/libz.dylib` with?

thecodeguyexamples, 2023-01-31:
### Update:

I have resolved the `zlib` issue by installing it wit MacPorts. However I am now getting the error below:

> :-1: error: Undefined symbols for architecture x86_64:
  "_git_annotated_commit_free", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_annotated_commit_lookup", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_checkout_tree", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_commit_create_v", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_commit_free", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_commit_lookup", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_index_free", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_index_write_tree", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_libgit2_init", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
      PartsChecker::getSha(QString const&) in partschecker.o
      PartsChecker::updateParts(QString const&, QString const&, PartsCheckerUpdateInterface*) in partschecker.o
  "_git_libgit2_shutdown", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
      PartsChecker::getSha(QString const&) in partschecker.o
      PartsChecker::updateParts(QString const&, QString const&, PartsCheckerUpdateInterface*) in partschecker.o
  "_git_merge", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_merge_analysis", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_object_free", referenced from:
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
  "_git_object_lookup", referenced from:
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
  "_git_oid_fmt", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
      PartsChecker::getSha(QString const&) in partschecker.o
  "_git_oid_fromstr", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_oid_tostr_s", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
  "_git_reference_free", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reference_lookup", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reference_name", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reference_name_to_id", referenced from:
      PartsChecker::newPartsAvailable(QString const&, QString const&, bool, QString&, PartsCheckerResult&) in partschecker.o
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
      PartsChecker::getSha(QString const&) in partschecker.o
  "_git_reference_resolve", referenced from:
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
  "_git_reference_set_target", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reference_shorthand", referenced from:
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
  "_git_reference_target", referenced from:
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reference_type", referenced from:
[... 30 lines cut ...]
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
      PartsChecker::getSha(QString const&) in partschecker.o
      PartsChecker::updateParts(QString const&, QString const&, PartsCheckerUpdateInterface*) in partschecker.o
  "_git_repository_state_cleanup", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_reset", referenced from:
      PartsChecker::cleanRepo(QString const&, PartsCheckerResult const&) in partschecker.o
[... cut ...]
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
  "_git_status_list_new", referenced from:
      PartsChecker::checkIfClean(QString const&, QString const&, git_repository*, PartsCheckerResult&) in partschecker.o
  "_git_tree_free", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o
  "_git_tree_lookup", referenced from:
      PartsChecker::doMerge(git_repository*, QString const&) in partschecker.o

Does anyone know why this is? For reference, I am using a 2020 M1 MacBook Air.

thenickdude, 2023-01-31:
Looks like you're missing libgit2

thecodeguyexamples, 2023-01-31:
I do have libgit2 installed in the same folder as my Fritzing app. I looked over the libgit2 docs and it looks like everything is set up right, is there a specific command I need to add to my `PATH` variable to make the library work?

closed 2024-02-13 by KjellMorgenstern